realsenseai/librealsense #4180: No Devices Found LabVIEW Windows 7

iggy12345, 2019-06-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.10.03.00) | 
| Operating System & Version |   {Win 7)  | 
| Platform                                 | PC  |
| SDK Version                            |  (2.11.0) | 
| Language                            |  { labview }                          | 

### Issue Description
No matter which port I plug my camera into, I cannot get the `Realsense_HelloWorld_Depth_and_Color.vi` to find my camera and I don't know what's going on.  I was able to narrow it down to `rs3 get device count2.vi` I can see a non-zero device list going in, but I get nothing but zeros out.

The realsense viewer is able to find my camera, so what's up with LabVIEW?

MartyG-RealSense, 2019-06-10:
The default setup of LabVIEW is for the 64-bit Windows 10 version of realsense.dll.  The wrapper's documentation front-page says that it can be made compatible with other Windows versions by compiling the SDK for that version and then substituting the default realsense2.dll in the location used by the LabVIEW demos for a dll from the older Windows version (renaming it to realsense2.dll before substituting it in if it isn't called that already)

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

iggy12345, 2019-06-10:
Pointing the DLL path in LabVIEW to `C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\bin\x86\realsense2.dll` worked!

MartyG-RealSense, 2019-06-10:
Awesome news, I'm very glad you found an approach that solved your problem!  :)

closed 2019-06-10 by iggy12345