realsenseai/librealsense #4181: Segmentation fault after updating firmware

LalitPradhan, 2019-06-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D420  | 
| Firmware Version                   | 5.11.4.0 | 
| Operating System & Version |   Ubuntu16  | 
| Kernel Version (Linux Only)    |  (4.15.0-50-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.22.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description
I installed via the process mentioned here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. 

On running realsense-viewer i kept getting messages to update the firmware to >5.10.

I followed the process mentioned here: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

Here is the output of my firmware update process:
$ intel-realsense-dfu -b 002 -d 003 -f -i /home/hansolo/Downloads/IntelD420/dkms/D400_Series_Development_FW_5_11_4/Signed_Image_UVC_5_11_4_0.bin 

 - Intel® RealSense™ DFU Tool v1.1.0-

Device busnum = 2
Device devnum = 3
FW file path = /home/hansolo/Downloads/IntelD420/dkms/D400_Series_Development_FW_5_11_4/Signed_Image_UVC_5_11_4_0.bin
FW Version of .bin File   = 5.11.4.0
FW version on device = 5.9.2.0
MM FW Version = 255.255.255.255
FW update required...
Updating FW...
DFU FW version in file = 5.11.4.0
DFU_version = 60
DFU_isLocked = 0
FW_highestVersion = 0.0.0.0
FW_lastVersion = 0.0.0.0

Percentage done: 99
Running post download processes...
.
Post download processes done.

Done!

When I run realsense-viewer with the camera connected, the application closes with this as the output on the terminal:
 10/06 17:43:14,205 WARNING [139687942130304] (synthetic-stream-gl.cpp:79) Initializing rendering, GLSL=1
 10/06 17:43:14,205 WARNING [139687942130304] (synthetic-stream-gl.cpp:88)  0 GPU objects initialized
 10/06 17:43:14,205 WARNING [139687942130304] (synthetic-stream-gl.cpp:131) Initializing processing, GLSL=1
 10/06 17:43:14,287 WARNING [139687942130304] (synthetic-stream-gl.cpp:145)  0 GPU objects initialized
**Segmentation fault (core dumped)**

my dmesg logs, lsusb logs, udev logs all detect the device.

On running 
$intel-realsense-dfu -p, the output is:

 - Intel® RealSense™ DFU Tool v1.1.0-

**Could not find device in opmode!
Device currently updating auxilliary modules, please try again later**

Can you please help me regarding this issue. I tried setting the $ export LRS_LOG_LEVEL="<Log Level>" to all possible values but end up getting a 0 byte file.

ev-mp, 2019-06-11:
@LalitPradhan hello, you've mentioned that you're using D420 model -
What is the output of  `rs-enumerate-devices -c -o -m` ?
Also, did you try to run other demos, such as `rs-depth` , `rs-capture` and what were the results.

Providing the librealsense logs is required to have a better understanding of your situation.
So you can run `realsense-viewer`, redirect and reconfigure its log to a file with the "settings" button in the upper right. Then connect the device to PC to collect the logs without compiling from sources.

LalitPradhan, 2019-06-12:
@ev-mp 

$ rs-enumerate-devices
Segmentation fault (core dumped)

$ rs-depth
Segmentation fault (core dumped)

$ rs-capture
Segmentation fault (core dumped)

I changed the settings as mentioned above to log, the log file has no content on seg fault.

Cheer! 
Lalit

ev-mp, 2019-06-13:
@LalitPradhan , can you locally compile and run `rs-enumerate-devices`, replacing this line
>log_to_console(RS2_LOG_SEVERITY_~~ERROR~~ **DEBUG**);

It should generate the initial data to analyze.

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@LalitPradhan Any update for this issue? Could you please provide the issue log as ev-mp suggested? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@LalitPradhan If no other questions, will close this ticket accordingly. Thanks!

closed 2019-07-15 by RealSenseCustomerSupport