realsenseai/librealsense #4183: Double t265 connection

stino78, 2019-06-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-06-11:
@stino78 hello,
For T265 there is a limitation that requires the sensors to be mapped to the first `context` created in code.
To make your code run you just need to reuse the first `context` when opening the second device:
>pipe2 = rs.pipeline(~~ctx2~~**ctx1**)

It seem that you can drop the second `context` altogether.
Note that this doesn't apply to the other Realsense devices -  D400 and SR300.

See more discussions on this topic in #3892, #3734, #3434, #3437, #3892 

P.S: The original snippet was re-formatted for readability

stino78, 2019-06-11:
Hi ev-mp, thanks for the quick feedback!

Sadly I get a similar error...

_11:04:50:536 [011440] [E] Device-0C40: State [ACTIVE_STATE] got event [ON_PLAYBACK_SET] ==> [ERROR_STATE]
11:04:50:536 [011440] [E] Device-0C40: Entered state [ERROR_STATE]
Traceback (most recent call last):
  File "double_t265_example.py", line 40, in <module>
    pipe2.start(cfg2)
RuntimeError: Failed to start TM2 camera_

I am looking at the other related discussion, but I could not find a solution.

Can you help me with this?

Thank you,

Agostino

dorodnic, 2019-06-12:
This will have to be resolved via a code fix, and not a trivial one, unfortunately. We are working on it, but it can take more time.

karry3775, 2020-01-14:
@stino78 I am facing the exact issue right now. I was using a ros package called realsense-ros to launch multiple cameras using (rs_multiple_devices.launch) but I guess that doesn't support t265 cameras. Were you able to extract any pose information from t265 using your code. Thanks

stino78, 2020-01-14:
Hi Kartik,

Intel released this code, that is working for me:
https://github.com/IntelRealSense/realsense-ros/files/3461264/more_devices_test.py.zip

I use it with 2 cameras on Win 10 and Ubuntu 18

Best,

Agostino

karry3775, 2020-01-14:
Thanks for the prompt reply . I will definitey check this out. Appreciate
your help.

karry3775, 2020-01-15:
@stino78 Do I need any other dependencies / files installed for this to work. Or is there a command line argument that I need to pass for connecting two devices. Currently just running the python file gives some syntax error :

File "more_devices_test.py", line 18
    def get_devices_serial_numbers(device_suffix:str='T265') -> [str]:
                                                ^
SyntaxError: invalid syntax

Please let me know if you faced similar issues.

stino78, 2020-01-15:
Mmm...if you installed the sdk and the Realsense Viewer is working
properly, you should be good to go

Did you check if you can import the librealsense2 properly?

What is tour level with Python? This looks like an error of cosing, maybe
due to python versioning. What version are you running? OS?

karry3775, 2020-01-15:
@stino78 I think python version was the issue now , I was running on Python 2.7
I ran with python 3 now.
It gives the following output, Were you also getting something similar:

[DEBUG] 15:48:22.070: (MainThread) T265 (943222110577) camera is ready.
[DEBUG] 15:48:22.111: (MainThread) T265 (845412111129) camera is ready.
Serial experiment --------------------------------------------------
serial_numbers ['943222110577', '845412111129']
0 0 943222110577 (0.0, 0.0, 0.0) 2020-01-15 15:48:22.537152
1 0 943222110577 (0.0, 0.0, 0.0) 2020-01-15 15:48:22.544937
0 1 845412111129 (0.0, 0.0, 0.0) 2020-01-15 15:48:22.642573
1 1 845412111129 (0.0, 0.0, 0.0) 2020-01-15 15:48:22.651024
Parallel experiment --------------------------------------------------
0 0 943222110577 (0.00025890962569974363, -0.002097188727930188, 0.00010264413140248507) 2020-01-15 15:48:22.651212
0 1 845412111129 (0.0, 0.0, 0.0) 2020-01-15 15:48:22.652803
1 0 943222110577 (0.00034518822212703526, -0.001163087203167379, 0.0002176408306695521) 2020-01-15 15:48:22.656179
1 1 845412111129 (-5.85176576350932e-06, -0.00012395156954880804, -4.087463821633719e-06) 2020-01-15 15:48:22.657700

stino78, 2020-01-15:
Honestly, I have no idea...I started months ago and I modofoed quite much
the code

The good thing with yours is that you can read the SN, which means that
everything is probably working fine

karry3775, 2020-01-17:
Thanks for your help I got it to work finally. I had to merge the code that
you sent me and extract_pose code on T265's documentation.
Greatly appreciate your assistance.

stino78, 2020-01-17:
Glad it worked!!

closed 2020-09-11 by RealSenseSupport