realsenseai/librealsense #4185: rs-motion Sample Question

giovannicairoli, 2019-06-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer 2.22) | 
| Operating System & Version |   Win 10                                                 | 
| Platform                                 | PC                                                          |
| SDK Version                            |   legacy 2.22.0                                       | 
| Language                            |  {C++ }                                                      | 

### Issue Description
Good morning everyone,

I'm new in programming and Intel Cameras applications.
For this reason I started exploiting the samples. 
Regarding the rs-motion Sample it works. But it is not very fast. Also when I added also the Depth part in order to calculate the distance. In this way I'm not able to estimate the angles Runtime.

After some experiments I noticed that the time delay changes in function of the configuration. See the code below:

	// Add streams of gyro and accelerometer to configuration
	cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 63);
	cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F, 200);
	// Add stream of Depth
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

Changing this parameters or deleting the Depth configuration the response time change.

I didn't find any documentation about how to configure the streams. Is there anyone that can help me?
Maybe my problem is not in the code that I posted before; I just noticed some changes.

I would like to thank you in advance for your help.

Have a nice day,

Giovanni

giovannicairoli, 2019-06-13:
@ev-mp Could you please Help me?

Thank you,

Regards,

Giovanni

ev-mp, 2019-06-13:
@giovannicairoli hello,
I'd assume that the "response time" you mention is the GUI responsiveness (?)
The `rs-motion` is intended to visualize the device orientation in space, so it is fundamentally bounded to the display's refresh rate (60Hz) where the opengl render context is invoked every 15ms or so. Adding `DEPTH` stream to the configuration would affect performance to some degree, specifically the Depth->RGB color mapping is CPU and memory-intensive. (BTW, the configuration you did is perfectly valid)

You can modify the example by disabling the v-sync, and get higher fps rates, but for production-grade code with high fps rates you should probably review the different methods the SDK offers:

There are (at least) three modes of frames acquisition:
1.	**Low-level “sensor” callbacks API** calls:
-	Minimum SDK-imposed overhead == minimum latency.
-	Each frame is handled with a separate callback  - “first come – first serve”.
-	All the synchronization stuff is left for the user side to implement.
-	Check `rs-data-collect` tool of the SDK that uses this method
2.	**Synchronous pipeline API** calls:
-	Blocking/Non-blocking calls which allow for “single-thread” exe model 
-	`Pipe.wait_for_frames/poll_for_frames` represent this method
-	The resulted framesets are (temporally) synchronized internally
-	Most of the SDK demos use this method (e.g. `rs-capture`)
3.	**Asynchronous pipeline API** calls:
-	Generates callbacks similar to the low-level API, but those callbacks provide synchronized framesets. Note that the synchronization affects UVC streams only (Depth,IR/Color/Fisheye)
-	In case of IMU frames – due to the high-fps rates these are not being synchronized intentionally and are served immediately
-	Depth/RGB/IR callbacks will be raised each time one of those streams is updated. But since the frameset always delivers a coherent snapshot of all configured streams, you should anticipate that the same frames will appear in more than one callback by design.
-	Check `rs-pose-and-image` demo

Related to https://github.com/IntelRealSense/librealsense/issues/3817#issuecomment-485663496

giovannicairoli, 2019-06-13:
Hi,

Thank you for your answer. So if I understood, it is not possible to put together to samples(Motion and Distance) without modifying something in the code and in the configuration?

I was hoped to merge to samples without adding more sophisticated code. I'm new in programming and I'd like to grow step by step. 

I will try to find why it is not working.

Thank you,

Giovanni

ev-mp, 2019-06-13:
Hi @giovannicairoli , the `rs-motion` example already employs method No:3 above optimized for performance. So it should be absolutely possible to get all streams at their declared FPS rate.
The frames are sent via separate callbacks so adding Depth stream would not affect the FPS of IMU streams. It can affect CPU utilization though, as mentioned above, and subsequently the UI response times, but that would mostly depend on how much processing performed on the Depth data.
To get a sense of what you can expect with the 3 streams turned on in terms of FPS/Responsiveness/CPU utilization  you can try `realsense-viewer`.

giovannicairoli, 2019-06-13:
Hi @ev-mp,

Thank you for your patience.
I will try again to useo Gyro, Accel and Depth together without changing anything in the configuration. From the realsense-viewer

giovannicairoli, 2019-06-14:
It works thankyou

closed 2019-06-14 by giovannicairoli