realsenseai/librealsense #4186: Building Matlab wrappers in OSx

guille31794, 2019-06-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2019-06-11:
Are you making a Feature Request for Mac OSX support to be added to the MATLAB wrapper, please?   Thanks!

guille31794, 2019-06-11:
Dear developers,
I'm trying to use a 435 camera in Mojave, through Matlab 2018a, but I'm unable to compile all the project.
I used cmake gui, and marked only the minimum default options + matlab binding to generate the project. After that, in terminal I type make in the realsense-sdk/build folder, and the project start compiling, but with lots of warnings.
When it reaches 51% and started compiling matlab part, it throw compile error, and finish it execution.
[Texto de Terminal.txt](https://github.com/IntelRealSense/librealsense/files/3276000/Texto.de.Terminal.txt)

Thanks in advance for any possible help.

guille31794, 2019-06-11:
> Are you making a Feature Request for Mac OSX support to be added to the MATLAB wrapper, please? Thanks!

No, sorry about the post. I was editing it. In op is all the info. Thanks.

lramati, 2019-06-11:
Could you try adding `#include <string>` to the top of factory.h? I'm not sure why this dependency seems to only be missing on Mac, but explicitly including it should solve your problems

MartyG-RealSense, 2019-06-11:
By 'matlab binding', do you mean the MATLAB wrapper for the Librealsense SDK, please?  If so, that wrapper is meant for use with Windows 10.

With the LabVIEW wrapper, which is also set to Windows 10 by default, it is possible to get it to work with other Windows versions by directing it to a realsense2.dll file that was built on the particular computer OS that you are using.  

https://github.com/IntelRealSense/librealsense/issues/4180

I do not know if something similar is possible with the MATLAB wrapper to get it to work with MacOS.  One of the Intel team members on this forum can hopefully provide useful advice about this.   Good luck!

guille31794, 2019-06-12:
I did it, and compiled factory.cpp, but not much more, It detects it lacks one file.

[Texto de Terminal.txt](https://github.com/IntelRealSense/librealsense/files/3280082/Texto.de.Terminal.txt)

lramati, 2019-06-12:
All the warnings in that output are normal, I'm not really sure where or
why the error is happening

guille31794, 2019-06-13:
No problem. I'll stand with windows for a couple of weeks. 
Hope an update for linux kernel 5.0.
Thank you for help and attention.

closed 2019-06-13 by guille31794