realsenseai/librealsense #4187: Instruction how to generate xcode project removed from installation_osx.md

matspetter, 2019-06-11:
In this commit:
https://github.com/IntelRealSense/librealsense/commit/90011e4c57b05d372f3a85feae784e8d0386f891#diff-dbe50ffd4ff0a727dcf0bbeaea04ce53

the instruction has been modified and no longer shows how to generate the xcode project files.

radfordi, 2019-06-11:
Does `cmake -G Xcode -H. -Bx cmake --build x`  not work for you?  It does for me.  There's nothing special about `XCode` support compared to `Unix Makefiles`, etc.

radfordi, 2019-06-11:
Never mind, you are right, @matspetter, that @matkatz shouldn't have removed the `-G XCode` from the section "Generate XCode project".  Clearly you don't need `-G XCode` for a makefile build, but you do need it for XCode.  Thanks for reporting.

dorodnic, 2019-10-16:
Restored, thank you

closed 2019-10-16 by dorodnic