realsenseai/librealsense #4188: pyrealsense2 object file not detected by python script

rijo11811, 2019-06-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

rijo11811, 2019-06-11:
@ev-mp Your take on this issue.

rijo11811, 2019-06-11:
@RealSense-Customer-Engineering 
Hello,
Would be great to have your take on this issue.

lramati, 2019-06-11:
Yes, the files you generated are for Python 3.6. The readme should have instructions on how to force Python 2 compilation. Please make sure you are indeed running Python 3 and not Python 2, which may be the default when running `python` from the command line

rijo11811, 2019-06-11:
@Iramati Hello

Thanks for the input. I am using python3.6 to run the script. No idea why it is not working.

malith1992, 2019-06-25:
What is the python script folder location you copied those files

rijo11811, 2019-06-26:
the issue is resolved. Lot of small mistakes from my end. The main issues was that link I created for the pyrealsense2 so was broken. It is working fine now.

closed 2019-06-26 by rijo11811

malith1992, 2019-07-10:
> The issue is resolved. Lot of small mistakes from my end. The main issues was that the link I created for the pyrealsense2 so was broken. It is working fine now.

@rijo11811 Can you please tell me how you create the link for the pyrealsense2. Because I have the same issue.

rijo11811, 2019-07-11:
@malith1992 We need to dive into the concept of links in Linux to figure out the issue. I used a work around. 
i copied my python script into the original location of pyrealsense2.so. PFB the same on my raspberrypi:

****@icets-ubuntu-mate:~/Desktop/librealsense-master/build/wrappers/python$

I had many pyrealsense2 so files here with names like "pybackend2.cpython-36m-aarch64-linux-gnu.so
" . Renamed the one which suits my system to pyrealsense2.so.