realsenseai/librealsense #4190: Latest firwmare links

NXTwoThou, 2019-06-11:
SDK 2.0 (build 2.23.0) recommends firmware 5.11.6.250+.  The two links provided are https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t which gives 5.11.4 and https://www.intelrealsense.com/developers/#firmware which is 5.11.1.100

When I search for D00 5.11.6 I get download locations from Softpedia and top4download, neither one I have any reason to trust.

Could we get updated links to find the firmware we're supposed to use?

AntoineWefit, 2019-06-11:
I kept the 5.11.6.200 firmware version if you need it.
[Signed_Image_UVC_5_11_6_200.zip](https://github.com/IntelRealSense/librealsense/files/3277193/Signed_Image_UVC_5_11_6_200.zip)

Although, I may be wrong but I think that this version was released **before** the current available versions. I downloaded it the 26/03/2019 so **it's not the latest version !** 
It looks like it has been removed since, I don't know what's going on with this firmware. Also, I'm not using it currently and I don't think it should be considered as a "stable version"... So you've been warned :)

NXTwoThou, 2019-06-11:
Thanks.  I trust a user on here a lot more than top4downlad.com :).  If firmware has been removed it might be best that I hold off until someone responds as to why build 2.23.0 recommends 5.11.6.250 and beyond but only 5.11.4 seems to be available publicly.

AntoineWefit, 2019-06-11:
> If firmware has been removed it might be best that I hold off until someone responds as to why build 2.23.0 recommends 5.11.6.250 and beyond but only 5.11.4 seems to be available publicly.

Yes indeed !

dorodnic, 2019-06-12:
I recommend to keep your current firmware or use latest from the official site.
It takes couple of days for new firmware to be uploaded, but new librealsense is not strictly coupled with latest firmware release.

NXTwoThou, 2019-06-12:
The release notes made it seem as though it required a minimum firmware version.  I'll keep checking back for the indicated firmware.

barnjamin, 2019-06-12:
The last 3 releases have listed 5.11.6.X for firmware but the latest on the firmware page is 5.11.1.100 from February.

barnjamin, 2019-06-13:
5.11.6.250 is available now

NXTwoThou, 2019-06-13:
Awesome!  Installing it now.  Thanks.

closed 2019-06-13 by NXTwoThou