realsenseai/librealsense #4191: 'rs-capture' fails to start and throws an interesting set of UVC errors

evercoast-smm, 2019-06-11:
Camera Model: D400
Firmware Version: 5.10.03
Operating System: Ubuntu
Kernel Version: 4.15.0-43-generic (we see the problem across a variety of kernel versions)
Platform: PC
SDK Version: 2.21.0
Language: C++
Segment: VR/AR

There are 18 cameras currently plugged into the system. We're confident in their available bandwidth and that they're being powered adequately.

We think that the driver is frequently left in a bad state. 'realsense-viewer' will start fine, but the simple example apps will not, hence we log a bug against the simplest example app.

We're wondering about the nature of these errors, their cause, impact, and remedy?

Dialog remains blank -- the following error messages are displayed in the shell:

_~/Realsense/librealsense-2.21.0/examples/capture$ rs-capture
 11/06 11:14:19,213 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video106
vid- 8086
pid- ad3
mi- 3
unique_id- 24-4.1.1-5
path- /sys/devices/pci0000:18/0000:18:01.0/0000:1f:00.0/0000:20:07.0/0000:23:00.0/usb24/24-4/24-4.1/24-4.1.1/24-4.1.1:1.3/video4linux/video106
susb specification- 320
metadata node-/dev/video107 was already been assigned 
 11/06 11:14:19,213 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
unique_id- 12-4-3
path- /sys/devices/pci0000:18/0000:18:00.0/0000:19:00.0/0000:1a:01.0/0000:1b:00.0/usb12/12-4/12-4:1.3/video4linux/video28
susb specification- 320
metadata node-/dev/video29 was already been assigned 
 11/06 11:14:19,213 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video48
vid- 8086
pid- ad3
mi- 3
unique_id- 32-4-3
path- /sys/devices/pci0000:8c/0000:8c:00.0/0000:8d:00.0/0000:8e:07.0/0000:91:00.0/usb32/32-4/32-4:1.3/video4linux/video48
susb specification- 320
metadata node-/dev/video49 was already been assigned 
 11/06 11:14:19,214 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video68
vid- 8086
pid- ad3
mi- 0
unique_id- 40-4-3
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:07.0/0000:9d:00.0/usb40/40-4/40-4:1.0/video4linux/video68
susb specification- 320
metadata node-/dev/video69 was already been assigned 
 11/06 11:14:19,214 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video88
vid- 8086
pid- ad3
mi- 3
unique_id- 30-4.1.1-5
path- /sys/devices/pci0000:8c/0000:8c:00.0/0000:8d:00.0/0000:8e:05.0/0000:90:00.0/usb30/30-4/30-4.1/30-4.1.1/30-4.1.1:1.3/video4linux/video88
susb specification- 320
metadata node-/dev/video89 was already been assigned 
 11/06 11:14:19,645 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video106
vid- 8086
pid- ad3
mi- 3
unique_id- 24-4.1.1-5
path- /sys/devices/pci0000:18/0000:18:01.0/0000:1f:00.0/0000:20:07.0/0000:23:00.0/usb24/24-4/24-4.1/24-4.1.1/24-4.1.1:1.3/video4linux/video106
susb specification- 320
metadata node-/dev/video107 was already been assigned 
 11/06 11:14:19,645 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
[... 123 lines cut ...]
metadata node-/dev/video29 was already been assigned 
 11/06 11:14:19,756 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video48
vid- 8086
pid- ad3
mi- 3
unique_id- 32-4-3
path- /sys/devices/pci0000:8c/0000:8c:00.0/0000:8d:00.0/0000:8e:07.0/0000:91:00.0/usb32/32-4/32-4:1.3/video4linux/video48
susb specification- 320
metadata node-/dev/video49 was already been assigned 
 11/06 11:14:19,756 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video68
vid- 8086
pid- ad3
mi- 0
unique_id- 40-4-3
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:07.0/0000:9d:00.0/usb40/40-4/40-4:1.0/video4linux/video68
susb specification- 320
metadata node-/dev/video69 was already been assigned 
 11/06 11:14:19,756 ERROR [140350922397568] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video88
vid- 8086
pid- ad3
mi- 3
unique_id- 30-4.1.1-5
path- /sys/devices/pci0000:8c/0000:8c:00.0/0000:8d:00.0/0000:8e:05.0/0000:90:00.0/usb30/30-4/30-4.1/30-4.1.1/30-4.1.1:1.3/video4linux/video88
susb specification- 320
metadata node-/dev/video89 was already been assigned_

ev-mp, 2019-06-11:
@evercoast-smm , what is the `dmesg` output when this error is encountered?

evercoast-smm, 2019-06-12:
"sudo dmesg -c" before running:

foo@foo:~$ dmesg
[  447.002981] ------------[ cut here ]------------
[  447.002983] Unknown pixelformat 0x00000000
[  447.003027] WARNING: CPU: 25 PID: 6273 at /build/linux-hwe-D1WrSl/linux-hwe-4.18.0/drivers/media/v4l2-core/v4l2-ioctl.c:1297 v4l_fill_fmtdesc+0xf51/0x12f0 [videodev]
[  447.003027] Modules linked in: thunderbolt nls_iso8859_1 intel_rapl skx_edac nvidia_uvm(OE) snd_hda_codec_hdmi nfit x86_pkg_temp_thermal intel_powerclamp coretemp nvidia_drm(POE) nvidia_modeset(POE) kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel hp_wmi pcbc wmi_bmof intel_wmi_thunderbolt sparse_keymap aesni_intel aes_x86_64 crypto_simd cryptd glue_helper nvidia(POE) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media intel_cstate snd_hda_codec_realtek snd_hda_codec_generic intel_rapl_perf joydev input_leds drm_kms_helper snd_hda_intel serio_raw snd_seq_midi drm snd_hda_codec ipmi_devintf snd_seq_midi_event snd_hda_core ipmi_msghandler snd_rawmidi fb_sys_fops snd_hwdep syscopyarea sysfillrect snd_pcm sysimgblt snd_seq snd_seq_device snd_timer
[  447.003060]  ucsi_acpi ioatdma snd mei_me typec_ucsi soundcore typec mei lpc_ich dca wmi mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 multipath linear hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid raid0 psmouse nvme nvme_core i40e e1000e ahci libahci
[  447.003086] CPU: 25 PID: 6273 Comm: rs-capture Tainted: P        W  OE     4.18.0-21-generic #22~18.04.1-Ubuntu
[  447.003087] Hardware name: HP HP Z8 G4 Workstation/81C7, BIOS P60 v01.72 11/29/2018
[  447.003090] RIP: 0010:v4l_fill_fmtdesc+0xf51/0x12f0 [videodev]
[  447.003091] Code: 53 4f 4e 58 0f 84 5c 02 00 00 3d 64 76 73 64 48 c7 c6 ee 28 9c c0 0f 84 71 f5 ff ff 89 c6 48 c7 c7 92 2a 9c c0 e8 4f 35 4e e1 <0f> 0b 80 7b 0c 00 0f 85 73 f5 ff ff 8b 43 2c 48 c7 c1 a0 1d 9c c0 
[  447.003110] RSP: 0018:ffff96aab64cbc48 EFLAGS: 00010282
[  447.003111] RAX: 0000000000000000 RBX: ffff96aab64cbd80 RCX: 0000000000000006
[  447.003112] RDX: 0000000000000007 RSI: 0000000000000092 RDI: ffff8b8dcf0564b0
[  447.003113] RBP: ffff96aab64cbc58 R08: 0000000000003d9c R09: 0000000000000004
[  447.003113] R10: ffff8b8dc17a9300 R11: 0000000000000001 R12: ffff8b8d5b6feb00
[  447.003114] R13: ffffffffc0a6f9a0 R14: ffff8b8dc17a9380 R15: ffffffffffffffed
[  447.003115] FS:  00007f6f72b3c780(0000) GS:ffff8b8dcf040000(0000) knlGS:0000000000000000
[  447.003116] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  447.003116] CR2: 00007f6f52ffcf78 CR3: 00000017b38e0001 CR4: 00000000007606e0
[  447.003117] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  447.003118] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  447.003118] PKRU: 55555554
[  447.003119] Call Trace:
[  447.003124]  v4l_enum_fmt+0x74/0x110 [videodev]
[  447.003127]  __video_do_ioctl+0x1ad/0x3c0 [videodev]
[  447.003133]  ? path_openat+0x608/0x1780
[  447.003139]  ? idr_alloc_u32+0xa4/0xe0
[  447.003142]  video_usercopy+0x1af/0x670 [videodev]
[  447.003145]  ? v4l_fill_fmtdesc+0x12f0/0x12f0 [videodev]
[  447.003147]  ? get_empty_filp+0xc9/0x1b0
[  447.003150]  video_ioctl2+0x15/0x20 [videodev]
[  447.003152]  v4l2_ioctl+0x49/0x50 [videodev]
[  447.003154]  do_vfs_ioctl+0xa8/0x630
[  447.003155]  ? putname+0x4c/0x60
[  447.003157]  ksys_ioctl+0x75/0x80
[  447.003158]  __x64_sys_ioctl+0x1a/0x20
[  447.003163]  do_syscall_64+0x5a/0x120
[  447.003168]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  447.003169] RIP: 0033:0x7f6f6fc965d7
[  447.003169] Code: b3 66 90 48 8b 05 b1 48 2d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 81 48 2d 00 f7 d8 64 89 01 48 
[  447.003187] RSP: 002b:00007fff48bc74b8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  447.003189] RAX: ffffffffffffffda RBX: 0000556d128dfa20 RCX: 00007f6f6fc965d7
[  447.003189] RDX: 00007fff48bc7670 RSI: 00000000c0405602 RDI: 000000000000001d
[  447.003190] RBP: 000000005a313620 R08: 0000000000000001 R09: 0000000000000008
[  447.003190] R10: 0000000000000015 R11: 0000000000000246 R12: 00007fff48bc76b0
[  447.003191] R13: 00007fff48bc7640 R14: 00000000c02c564a R15: 00007fff48bc78f0
[  447.003192] ---[ end trace aa2202bdae3aaf14 ]---
[  447.003363] ------------[ cut here ]------------
[  447.003363] Unknown pixelformat 0x00000000
[  447.003371] WARNING: CPU: 25 PID: 6273 at /build/linux-hwe-D1WrSl/linux-hwe-4.18.0/drivers/media/v4l2-core/v4l2-ioctl.c:1297 v4l_fill_fmtdesc+0xf51/0x12f0 [videodev]
[  447.003372] Modules linked in: thunderbolt nls_iso8859_1 intel_rapl skx_edac nvidia_uvm(OE) snd_hda_codec_hdmi nfit x86_pkg_temp_thermal intel_powerclamp coretemp nvidia_drm(POE) nvidia_modeset(POE) kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel hp_wmi pcbc wmi_bmof intel_wmi_thunderbolt sparse_keymap aesni_intel aes_x86_64 crypto_simd cryptd glue_helper nvidia(POE) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media intel_cstate snd_hda_codec_realtek snd_hda_codec_generic intel_rapl_perf joydev input_leds drm_kms_helper snd_hda_intel serio_raw snd_seq_midi drm snd_hda_codec ipmi_devintf snd_seq_midi_event snd_hda_core ipmi_msghandler snd_rawmidi fb_sys_fops snd_hwdep syscopyarea sysfillrect snd_pcm sysimgblt snd_seq snd_seq_device [... cut ...]
[  447.012659] RSP: 002b:00007fff48bc7b98 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  447.012660] RAX: ffffffffffffffda RBX: 0000556d127f2330 RCX: 00007f6f6fc965d7
[  447.012660] RDX: 00007fff48bc7ca0 RSI: 00000000c0405602 RDI: 0000000000000025
[  447.012661] RBP: 00007f6f72890668 R08: 0000000000000004 R09: 00007f6f6ff6bc40
[  447.012661] R10: 0000000000000007 R11: 0000000000000246 R12: 00007fff48bc7ca0
[  447.012662] R13: 00007f6f71b62a70 R14: 000001e000000280 R15: 00007fff48bc7e60
[  447.012665] ---[ end trace aa2202bdae3aaf19 ]---

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@evercoast-smm What's the hardware platform used? Did you apply kernel patch when install SDK?

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@evercoast-smm No response got from you. If no other questions about this issue, will close it accordingly. Thanks!

closed 2019-07-11 by RealSenseCustomerSupport