realsenseai/librealsense #4192: multicam HW Sync works but not perfectly

krishnashankar, 2019-06-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.10.6.0 | 
| Operating System & Version | Ubuntu 16.04| 
| Kernel Version (Linux Only)    |4.13.0-37-generic| 
| Platform                                 | PC  |
| SDK Version                            |2.16.5 | 
| Language                            |c++| 
| Segment			|Robot| 

### Issue Description

First of all, thank you for all of the amazing resources on getting cameras sync - between the whitepaper and the various issues other's have posted, we *almost* got our setup working. 

We have 4 realsenses and 2 Basler cameras in our setup that we'd like to sync together. We verify sync using an LED clock that can resolve down to ~1 ms. 

In order to trigger the baslers using the realsense master camera, we have a level-shifter on the sync signal.

We have been able to establish that using one of the realsenses as master, we are able to trigger the 2 Basler cameras successfully, and they capture at exactly the same instant. However, the realsenses seem to capture at slightly different times (error varies between 0 and 8 ms). All cameras in our experiments have an exposure time of 1ms.

It is clear that the trigger has an impact on the realsenses, as the difference in their capture is much greater when the sync is disabled.

We're verifying sync by looking at our LED clock in the the IR image (enabling IR and Depth streams) in Y8 format. 

I'd be happy to provide more code or information if it might help you tell us what we could fix. 
Thanks in advance!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

I think this issue might be related to ESD problem which is a known issue. Sometimes ESD causes false trigger over the sync cable and impacts the Slave camera. How does the sync cable look like? can you take a picture?

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

Hi @krishnashankar Do you need further help on this?
Thanks,

closed 2020-01-10 by RealSenseCustomerSupport