realsenseai/librealsense #4193: Could not initialize offscreen context!

Vaibhavjolly, 2019-06-12:
----------------------------------------------------------------------------------------------------
| Camera Model                       |D435 |  
| Operating System & Version | Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.4.0-143-generic |                                        | 
| Platform                                 | PC  |

Hi, 
I bought a intel realsense depth camera D435 and trying to install realsense .
I followed this : https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
When I am running realsense-viewer , I am getting the following error :

 Could not initialize offscreen context!

I checked the command : ldd $(which realsense-viewer) | grep 'realsense2' :
Output :  librealsense2-gl.so.2.23 => /usr/lib/x86_64-linux-gnu/librealsense2-gl.so.2.23 (0x00007f5278f01000)
	librealsense2.so.2.23 => /usr/lib/x86_64-linux-gnu/librealsense2.so.2.23 (0x00007f5277537000)

Thanks in advance for the help .
Vaibhav

dorodnic, 2019-06-12:
Please make sure GPU (OpenGL) drivers are correctly installed on your system. 
What type of GPU are you using?

Vaibhavjolly, 2019-06-12:
I am using this GPU -
product: GP102 [GeForce GTX 1080 Ti]

dorodnic, 2019-06-12:
Do you have nvidia drivers installed? 
This is not strictly related to RealSense, various textual examples will work, but for graphical examples you need your Linux system to be able to properly handle OpenGL.

Vaibhavjolly, 2019-06-14:
Yes, Its installed properly .

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@Vaibhavjolly Could you please double check if *glfw3*, *mesa* and *gtk* packages were installed? These packages are required according to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

    *glfw3*, *mesa* and *gtk* packages are required if you plan to build the SDK's OpenGl-enabled examples.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@Vaibhavjolly Any other questions for this issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@Vaibhavjolly If no other questions, will close it accordingly. Thanks!

closed 2019-07-15 by RealSenseCustomerSupport