realsenseai/librealsense #4196: How to connect my D415 with visual studio ?

nourbouabena, 2019-06-12:
Hi all,

I need to know how to encode my real sense camera D415 with visual studio in c sharp, I want to display the camera on my application in visual studio, can you help me to program or tell me how I can do?

Do you have any advice ?

thanks,

Nour BOUABENA

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@nourbouabena The link below might helps.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@nourbouabena You're welcome. Any other questions about this ticket?

RealSenseCustomerSupport, 2019-07-09:
----------------------------------------------

@nourbouabena If no other question will close the issue accordingly. Thanks!

closed 2019-07-09 by RealSenseCustomerSupport

hopaa, 2019-07-23:
Hi,
I built and ran .net examples successfully. But when I tried to create my own .net project, something went wrong. I can't figure it out!
I just copied and pasted the "cs-tutorial-4-software-dev" source files and add the required references to my WPF project. Whereas it was built successfully, when I tried to run it, this exception occurred:
"The type initializer for 'Intel.Realsense.Context' threw an exception."

I traced the code, when it tries to get an instance of a Pipeline, it calls a line of code that wants to get an instance of Context which throws the exception.

I mean this line of code throws the exception:
"using (var ctx = new Context())"  --> line 35 of Pipeline.cs

Any idea how to handle it?! 

Thanks in advance