realsenseai/librealsense #4197: Unit Testing - emulating a realsense device using `software_device`

callendorph, 2019-06-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | Doesn't matter | 
| Operating System & Version |   Ubuntu 16 - but applies to all  | 
| Kernel Version (Linux Only)    |  4.4.0 but applies to all                            | 
| Platform                                 | PC  |
| SDK Version                            |  master                         | 
| Language                            |  C / C++ / python but applies to all                          | 
| Segment			|  Unit Testing                          | 

### Issue Description

I would like to be able to write a pure software unit test of our code that interfaces to the realsense camera via pyrealsense2. We've been very successful at using the `uhid` and `vcan` modules to write emulated devices for unit testing of hardware in our project and I'd like to do something similar for the realsense camera. From what I can tell, the `software_device` class should allow me to do what I want to do. Unfortunately, I've run into an issue that is making this difficult. 

#### Steps

1.  I've confirmed that I can use the realsense hardware camera with librealsense and the realsense viewer application - everything works fine.
1.  I updated the `pybind` interface in `wrappers/python/python.cpp` to include the `software_device` and `software_sensor` classes.
    1.  I'm referencing the classes defined in the external API not the classes referenced in `src/software-device.h`
2.  Re-built pybind targeting my local python executable. 
3.  Loaded the module and started a python session: 

```python
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> dev = rs.software_device()
>>> sensor = dev.add_sensor("depth")
>>> ctx = rs.context()
>>> dev.add_to(ctx)
>>> list(ctx.query_devices())
[Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: info Serial Number not supported by the device!
>>> dev.supports(rs.camera_info.serial_number)
False

```

So at this point, I'm stuck. I can't seem to find a way to add the serial number to my `software_device` given the current external API. I think I want to use the `info_container::register_info` method which is inherited as a base class of the `software_device` defined in `src/software-device.h` - but this interface is not accessible via any of the external APIs. 

#### Questions

1.  What is the status of `software_device`? Is this in active development?
2.  Is there a reason why the `software_device` interface is restricted in this manner?
3.  Can you think of anything else that is going to block this approach that I should watch out for?
4.  Is there another way to do this that I'm overlooking? 
2.  If I modify the API to add the functionality necessary to implement an emulated camera, will this be considered for inclusion?

dorodnic, 2019-06-13:
Hi @callendorph 

> What is the status of software_device? Is this in active development?

At the moment it is not, for python wrapper current task that was completed was sphynx documentation and the next task I planned was adding more comprehensive examples.

> Is there a reason why the software_device interface is restricted in this manner?

Mainly because it is relatively corner-case functionality. We add APIs based on community interest, in the C# wrapper, for instance, software device is available, because some users needed it. 

> Can you think of anything else that is going to block this approach that I should watch out for? Is there another way to do this that I'm overlooking?

pybind11 is very nice tool for wrapping C++, it should be pretty straightforward. I'm a bit concerned about handling the "frame deleter" but it should be possible. @lramati should have more insights

> If I modify the API to add the functionality necessary to implement an emulated camera, will this be considered for inclusion?

Of course, we appreciate and encourage 3rd party contributions.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @callendorph

Can this be closed out now?  Your PR associated with this was closed already as well.

closed 2020-01-27 by callendorph

brunovollmer, 2021-02-04:
@callendorph Could you supply an example of how you implemented the unit test with an emulated device?