realsenseai/librealsense #4198: C# Match RGB pixel to depth pixel

aeric90, 2019-06-12:
Required Info |  
-- | --
Camera Model | D435
Firmware Version | 05.11.04.00
Operating System & Version | Win 10
Platform | PC Unity
SDK Version | 2.0
Language | C# 
Segment | AR

I have a program already set up in order to periodically poll depth data and process the information. It generates a texture of in a particular distance range and gives a general idea of their volume. This isn't in real-time. About every 1.5 to 2 seconds. However, I also need to occasionally check the RGB camera for the location of a particular colour from a projected image in the camera's field of view and then be able to get the depth information for the same location. I'm not sure what the best way of doing this is, and I haven't found very many C# implementations that do what I'm trying to do.

The help is very much appreciated!

ogoshen, 2019-06-18:
Sounds like you might be looking for the `Intel.RealSense.Align` processing block, you can use it to do [frame-alignment](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#frame-alignment).

RealSenseCustomerSupport, 2019-07-30:
----------------------------------------------

Hi aeric90,
Do you still need help on this?

aeric90, 2019-07-31:
Hello,

Yes. I was unable to find a suitable example of using Intel.RealSense.Align in C# so I'm using a manual alignment/calibration process currently which has some inaccuracies. Any additional assistance would be very helpful.

Cheers,
Eric

RealSenseCustomerSupport, 2019-08-09:
----------------------------------------------

The modified sample code contributed in the following thread may be useful in this case:
https://github.com/IntelRealSense/librealsense/issues/4523
Thank you.

closed 2019-08-20 by RealSenseCustomerSupport