realsenseai/librealsense #4201: Distance Array from Depth Frame in C# (unity)

mbampi, 2019-06-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |  Win (8.1/10) | 
| Platform                                 | PC |
| SDK Version                            |   2.0                         | 
| Language                            |  C#/unity                           | 

### Issue Description

How to get the Depth Frame data and convert it to a short(uint16) array. 
And then how to get the distance information (meters or cm) from the array?
Using C# please.

```c#
public static double[][] GetDistanceArray(DepthFrame depthFrame, float depthScale)
    {
        short[] input = new short[depthFrame.Stride * depthFrame.Height];
        Marshal.Copy(depthFrame.Data, input, 0, depthFrame.Stride * depthFrame.Height);

        for (int i = 0; i <  #input.Length; i += 2)
        {
            double distance = input[i] * depthScale; // meters
        }
    }
```

I tried doing something similar to the code (up^) but its not working.
Any help would be great.

Thank you!

ogoshen, 2019-06-18:
Couple of issues with the code:
1. Raw depth data in Z16 format is in **unsigned** shorts, `Marshal.Copy` has no overload for that type, use the `VideoFrame.CopyTo<T>` method.
2.  The `Stride` property is in bytes, when using ushorts the array size is width &times; height

```cs
var depthData = new ushort[depthFrame.Width * depthFrame.Height];
depthFrame.CopyTo(depthData);
```

Also see, [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#unsafe-direct-access) for accessing the data without copying in an unsafe context.

mbampi, 2019-06-25:
Thank you very much @ogoshen  👍

closed 2019-06-25 by mbampi

cristianbercu, 2021-01-28:
Much appreciated @ogoshen! 🌹