realsenseai/librealsense #4202: Align depth stream to external camera stream

abuzaina, 2019-06-12:
I have a D410 on a Jetson and an external 4K camera (MIPI interface). I can capture depth frames from the D410 and RGB frames from the 4K camera concurrent and in real time.
I would like to align the depth frame to the color frame of the external camera, similr to the rs-align sample. 
Is it possible to create a synthetic stream and pass the intrinsics and  extrinsics of the external camera to it ? If so how to do it?

Thanks

ev-mp, 2019-06-16:
@abuzaina hello, this scenario is handled with the `software-device` class that the SDK provides.
Using this construct allows to bind and handle otherwise non-related sensors with a mockup "device", like in your case. Other cases include synchronizing depth streams of multiple depth cameras to generate a panoramic point cloud.
The `software-device` demo shows how to use this class, including building up the virtual sensors, generate the "software sensor" options, intrinsic  and inter-stream extrinsic and then sending "user-defined" frames to Librealsense pipe - the core blocks to implement your use-case :
https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

This class is extensively utilized internally for unit-testing, see usages [here](https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-post-processing.cpp#L220-L265) and [here]()

ev-mp, 2019-06-17:
@abuzaina hello.
One thing that I didn't mention which is, imo the real challenge in this scenario is _obtaining the intrinsic matrix for the external sensor, and the extrinsic matrix between D410 and the external sensor_.
- The camera intrinsic matrix could probably be provided by the external sensor/OEM API or spec.
  It should include the focal and principal points, the distortion model and its coefficients. You'll have to [check that Librealsense  supports the distortion model](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L25-L119) exhibited by the 3rd-party sensor.
- The extrinsic matrix between D400 and the external sensor is the one that you'll have to generate on your own.

abuzaina, 2019-06-24:
hi @ev-mp ,
Thank you for your reply, the second usage link in your first reply does not go anywhere. I would appreciate if you send it again.
I am still not sure how to inject the depth frames from the D410 and the RGB frames from the camera to the software device, are there any examples of it ?

RealSenseCustomerSupport, 2019-07-19:
----------------------------------------------

@abuzaina Are you asking about the links in "This class is extensively utilized internally for unit-testing, see usages here and here"?

Please find them as blow.
https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-post-processing.cpp#L220-L265
https://github.com/IntelRealSense/librealsense/issues/4202

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

@abuzaina Any other questions about this? Thanks!

jb455, 2019-07-25:
I'm not OP but I have a related question. I had a similar thing to this (using software device to align depth stream with other camera) working with an older version of the SDK (2.14) and I have recently updated to the latest SDK (2.24) and now when I align, the depth stream is blank (all zeroes). Has there been a regression or is there something else I now need to do to get it working? Attached is modified Window.xaml.cs from cs-tutorial-4-software-dev to reproduce the issue.
[Window.xaml.cs.txt](https://github.com/IntelRealSense/librealsense/files/3430787/Window.xaml.cs.txt)

ev-mp, 2019-07-29:
@jb455 , please open a separate ticket for the C# issue

RealSenseCustomerSupport, 2019-08-16:
----------------------------------------------

@abuzaina Any other questions about this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-08-20:
----------------------------------------------

@abuzaina Sorry that we didn't get feedback from you for weeks. Will close this issue accordingly. Please feel free to create new ticket if you still have issues/questions. Thanks!

closed 2019-09-03 by RealSenseCustomerSupport