realsenseai/librealsense #4206: Updating D400 firmware from Windows command line

jb455, 2019-06-13:
Hi,
Is it possible to automatically update the firmware on a D400 device by command line in Windows?
When users start our software, we want to check the firmware on the attached device and update it if necessary by calling a batch file or similar. Our users will likely be non-techy so it's not reasonable to ask them to use the current dfu tool manually, and we won't necessarily get hold of each device before it goes to customers to update the firmware ourselves.
Our users are seeing lots of camera stability issues which should be fixed by firmware updates but most of them are using very old firmwares.
It appears the Linux version of the updater does support command line arguments, is it possible to port this over to Windows?

Thanks

dorodnic, 2019-06-13:
Hi @jb455 
I'm not sure about the current util, but we are working on new version that will support this. It will be available in short while

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@jb455 The latest SDK v2.24 now can support to new firmware update tool for cross platforms including windows: https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update
Please have a try and feedback. Thanks!

jb455, 2019-07-11:
Looks perfect, thanks!

closed 2019-07-11 by jb455