realsenseai/librealsense #4207: Examples on combining T265 & T435?

antoan, 2019-06-13:
I came across the talk below, which featured a demo combining the two cameras for autonomous navigation. Was if wondering source for this will be released at some point or if there are any examples/guidance/starting point on combining those two sensors in particular.
https://www.youtube.com/watch?v=62vm0_RZ1nU&t=493s

MartyG-RealSense, 2019-06-13:
Intel held an online webinar session yesterday (June 12 2019) on depth + tracking.  It should be released on the RealSense events archive site at some point in the near future.

https://www.intelrealsense.com/webinars-and-events/

In the meantime, the presenters provided a couple of links to resources for using tracking and depth together.

https://support.intelrealsense.com/hc/en-us/community/posts/360031819973

antoan, 2019-06-14:
Really appreciate it, thanks.

closed 2019-06-14 by antoan