realsenseai/librealsense #4209: Unity sample scene - SLAM - Not working

kouglov, 2019-06-13:
Hello, I open this issue because I have the same problems each time I try to run the SLAM sample scene from the UnityPackage.

Just tested it on a fresh install, everything's up to date.
I'm on Win10, Unity 2019, VS2019 and last updates of RealSense SDK.
Repo was cloned with the last version.
[image]

As you can see, I have the same error some other people did, but I didn't find any solutions as the issues were closed by the people who opened it, or they didn't give any clue on how they resolved it.

The complete stacktrace is : 

`ExternalException: rs2_pipeline_start_with_config(pipe:0000018DC887F3A0, config:0000018DC887F920)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <39ff202c67d64a63acbba4ae542e7167>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)`

As you can see on the next capture, device work properly.
In fact, the others unity scenes works properly.
Only the SLAM one bug.

[image]

Maybe I can also add that when you unpack the unitypackage in Unity 2019 there's a bunch of errors on UI scripts. 
[image]

Those scripts are far from essentials to launch the realsense camera, but I suggest the devteam take a look at that too.

ogoshen, 2019-06-18:
You're trying to run the 6DOF V-SLAM sample scene in Unity which requires the [Intel® RealSense™ Tracking Camera T265](https://www.intelrealsense.com/tracking-camera-t265/).

kouglov, 2019-06-21:
So, Intel developped devkit cameras capable of slam only with realtime 3D capture, and meshing, used in the R200 and SR300 cameras, but removed it in the V2 SDK ?
I mean, we have a camera capable of getting a point cloud in realtime, and suddenly we're not able again to do 3D mesh reconstruct and pseudo-slam in realtime with the new devkit ?
What the actual hell is that ? 

I've used the R200 many times with this kind of feature.
It was not perfect but very usable. I stil have the 3D meshes generated with it today, it worked, and it's the same exact technology in the V2 cameras.
I don't need a specialised slam camera or a depth camera with a rotation sensor because I know a simple depth camera is fully capable of creating a 3D mesh in realtime and recalibrating in realtime onto it.

I find this very strange, if someone from intel read this, how can you do this to developpers ?
I mean, the camera is fully capable of doing it, but you obviously deleted the software capacity to do it.
I don't understand this kind of positionning.
You're selling cameras that does less things than 3 years ago.
What kind of disapointment is that ?

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

Hi @kouglov,

You can refer to below page to enable SLAM feature with the open source although it's not running on Unity.
http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping

closed 2019-09-20 by RealSenseCustomerSupport