realsenseai/librealsense #4210: How to save pointcloud as .pcl file in python ?

Gazi13, 2019-06-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I want to save pointcloud as a " .pcl  " file in python but i couldn't find any way. Is is possible in python ? 

i just find this
`
pc = rs.pointcloud()
points = rs.points()
...
pc.map_to(color_frame)
points = pc.calculate(depth_frame)
points.export_to_ply("./ply_file.ply", color_frame)
`

MartyG-RealSense, 2019-06-14:
I hope that the discussion in the link below will be useful to you.

https://github.com/IntelRealSense/librealsense/issues/2483

In that discussion, it makes reference to how a RealSense community member recommends a solution called **python-pcl**

https://github.com/strawlab/python-pcl

Gazi13, 2019-06-18:
Thank you for your response.<br />

This **python-pcl** did not work for me but i did with [pclpy](https://pypi.org/project/pclpy/)
My solution 

```
import pclpy 
from pclpy import pcl
...
pc = rs.pointcloud()
points = rs.points()

...

depth_intrinsics = rs.video_stream_profile(depth_frame.profile).get_intrinsics()
w, h = depth_intrinsics.width, depth_intrinsics.height
out = np.empty((h, w, 3), dtype=np.uint8)
out.fill(0)
points = pc.calculate(depth_frame)

v, t = points.get_vertices(), points.get_texture_coordinates()
verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz
texcoords = np.asanyarray(t).view(np.float32).reshape(-1, 2)  # uv
            
pc2 = pcl.PointCloud.PointXYZ(verts)
writer = pcl.io.PCDWriter()
writer.writeBinary("./PointCloud.pcd",pc2)

```

MartyG-RealSense, 2019-06-18:
Thanks for sharing your solution!

dorodnic, 2019-06-18:
PCD is a format closely related to PCL, so it makes sense to use one of the PCL wrappers to do it. 
We are not planning to add this capability directly to librealsense.
Given a solution was found, I'm closing this one. @Gazi13 let us know if you need any more information.

closed 2019-06-18 by dorodnic