realsenseai/librealsense #4211: Align Color to Depth performance improvement tip

pavloblindnology, 2019-06-14:
When aligning color frame to depth frame there is no need in projecting the depth pixel onto the color image region using the corners, as it's done here (as well as in sse and cuda versions) 
https://github.com/IntelRealSense/librealsense/blob/d6f6be84b46190c8c84c95f6ac279d239320fcda/src/proc/align.cpp#L52
Because only the last color pixel value inside the region is assigned to aligned pixel
https://github.com/IntelRealSense/librealsense/blob/d6f6be84b46190c8c84c95f6ac279d239320fcda/src/proc/align.cpp#L99
Instead, it's enough to project the center of depth pixel only.

GucciPrada, 2019-06-14:
well, I guess the author wanna support align two images of different resolution or even of different ratios of width and height.

pavloblindnology, 2019-06-14:
It doesn't matter. Your depth pixel [u,v] is projected onto color image region [y0,y1]x[x0,x2], then you iterate over the region assigning value of every single color pixel to THE SAME aligned pixel [u,v]. What's the point in iterating then?

GucciPrada, 2019-06-15:
I'm not quite sure about the reason why the author did the iterating. I just could tell my understanding. 

I guess there is something that does matter. Like you said,  the [y0 y1]~[x0 x1] region needs to be filled by the same value but not left  empty or filled with other not existing values.

And actually, while aligning the depth image to another viewport,  there might exist some occlusion to take into consideration. So once the whole image alignment is done,  you might find out that  the region is not be filled with the same value any more, cuz' some pixels in the region were overlaid by pixels of neighboring region.

hope I'm not missing something ; P

pavloblindnology, 2019-06-18:
In fact, you're describing the reverse process - Depth to Color alignment, where points mentioned by you really matter.
On the contrary, during Color to Depth alignment [x0,x1]x[y0,y1] region of color image is projected on to a single depth pixel. So, after iterating over the region, the last [x1,y1] pixel's value is assigned to depth pixel, which makes iterating meaningless.

GucciPrada, 2019-06-19:
Yes, I think you are right.  
In my opinion, aligning color image of one viewpoint to the depth image of another different viewpoint is an ill-posed problem. It will cause some terrible artifacts in the aligned color image if the geometries are very complex (also depends on how different the two viewpoints are). So I never use it.

pavloblindnology, 2019-06-19:
Artifacts are another problem. Of course there are some. But sometimes you just need to do alignment to get RGBD.

dorodnic, 2019-06-19:
Hi @pavloblindnology 
Appreciate the suggestion. There is no singular **author** to this code, it was contributed to by many developers. Do you have any data on the performance impact of what you are suggesting? There is fairly significant test-set for alignment validation in the unit-tests (since it is very easy to break with all the combinations of RGB and depth resolutions) in-case you'd consider contributing your optimization to the project.

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@pavloblindnology Any update for the questions from Dorodnic? Looking forward to your reply. Thanks!

pavloblindnology, 2019-07-12:
Hi @dorodnic
No, I don't have any data on performance, but presume it should be significant since inner loop will be totally eliminated.

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@pavloblindnology Could you please submit one PR for your suggestion? Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@pavloblindnology Any update about this? Thanks!

pavloblindnology, 2019-10-01:
@RealSenseCustomerSupport Well, unfortunately, don't have time to contribute.

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@pavloblindnology That's all right. Thanks for your suggestions!

closed 2019-10-31 by RealSenseCustomerSupport