realsenseai/librealsense #4212: Do D415 camera have a mechanism to detect the disconnection and connection of camera from usb.

itachi134999, 2019-06-14:
For the real time, i am calling get all device query in every API calls which is taking around 500ms. Do librealsense have a mechanism to detect the connection and disconnection of the D series camera from USB ?

MartyG-RealSense, 2019-06-14:
The discussion linked to below may give you useful insights if you have not seen it already.

https://github.com/IntelRealSense/librealsense/issues/931

itachi134999, 2019-06-19:
def test():
    print('tested')

rs.context().set_devices_changed_callback(test())

This is not working for camera connection and disconnection, test() function is only called when at the initialization of set_devices_changed_callback.

lramati, 2019-06-19:
There are a couple errors in your python code there.
First of all, by not saving the context object, you're allowing it to be destroyed, at which point even if you had registered the callback correctly, it would not longer be listening.

Furthermore, by using `set_devices_changed_callback(test())`, you are feeding `set_devices_changed_callback()` the results of running the `test()` function (in this case `None`), not the function itself.
The correct way to handle this is:
```python
ctx = rs.context()
ctx.set_devices_changed_callback(test) # Note test isn't called
```

itachi134999, 2019-06-19:
I tried in that way but it is still not calling the test function at the time of disconnection or connection

lramati, 2019-06-19:
It might be a problem with the function signature?

itachi134999, 2019-06-20:
This is the code which i wrote:

rs_context = rs.context()

def test():
    print('dsfds')

rs_context.set_devices_changed_callback(test)

But this connection disconnect callback is not working

itachi134999, 2019-06-25:
Can anyone tell me what is the problem here, why callback is not working at the time of USB connection or disconnection.

Is the problem in the code ?

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

Please try this code.

    import pyrealsense2 as rs2
    import time

    current_devices = []

    def on_devices_changed(info):
        global current_devices
        devs = info.get_new_devices()
        print("on_devices_changed called new_devices:", devs.size())
        for dev in devs:
            current_devices.append(dev)

        for dev in current_devices:
            print("current:", dev, " added:", info.was_added(dev), " removed:", info.was_removed(dev))

    def main():
        ctx = rs2.context()
        ctx.set_devices_changed_callback(on_devices_changed)
        print("Please connect/disconnect your RealSense camera.")
        time.sleep(10)
        print("Done.")

    if __name__ == '__main__':
        main()

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@itachi134999 Did you get chance to try? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

I think your issue has been solved.
If not, please reopen this ticket, and add more details.
Thanks.

closed 2020-03-20 by RealSenseCustomerSupport