realsenseai/librealsense #4214: librealsense Android 2.23.0: Crash in rs2_delete_sensor

xtrawurst, 2019-06-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | any | 
| Firmware Version                   | any | 
| Operating System & Version |  Android 8.0 | 
| SDK Version                            |  2.23.0 | 
| Language                            |  C++  |

### Issue Description
Crash shortly after RS context creation.

Running `ndk-stack` with the symbol-ed SO file and the crash dump revealed
```
********** Crash dump: **********
Build fingerprint: 'samsung/dreamltexx/dreamlte:8.0.0/R16NW/G950FXXU4CRI5:user/release-keys'
pid: 27223, tid: 27223, name: t3d.internaldev  >>> de.dotproduct.dot3d.internaldev <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xffffff
Stack frame #00 pc 0036f25c  /data/app/de.dotproduct.dot3d.internaldev-rjrKaX80zfiwa4A0FJmyKw==/lib/arm/librealsense2.so (rs2_delete_sensor+27): Routine rs2_delete_sensor at /Users/xyz/code/librealsense/package/src/rs.cpp:297
Stack frame #01 pc 00025cef  /data/app/de.dotproduct.dot3d.internaldev-rjrKaX80zfiwa4A0FJmyKw==/oat/arm/base.odex (offset 0x1f000)
```

xtrawurst, 2019-06-28:
@RealSense-Customer-Engineering @dperox @matkatz 

The Android part of librealsense often maps C pointers to Java `long`s to be used as handles. This is problematic when targeting 32-bit targets on Android (such as the still popular armeabi-v7a) as the pointer width (32 bit) doesn't match to the `long` width (64-Bit) and so handles may contain garbage in half of the `long`.

Attached is a patch file that fixes these issues. It may not be complete, i.e. there may be other parts of the code where bad casts are performed.

[0001-Android-Fixed-crashes-related-to-32-bit-pointers-not.txt](https://github.com/IntelRealSense/librealsense/files/3340103/0001-Android-Fixed-crashes-related-to-32-bit-pointers-not.txt)

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@xtrawurst Thanks very much for the patch! We'll look into this and update you later. Thanks!

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@xtrawurst Could you please submit one PR for this? Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@xtrawurst Did you get chance to submit the PR? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@xtrawurs Any progress for the PR? Thanks!

xtrawurst, 2019-11-08:
@RealSense-Customer-Engineering I'd do a PR but frankly I don't know how to do this through GitHub. The attached patch text file should contain everything that is needed though.

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

@xtrawurst This is the PR submit guideline https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md  Welcome your contribution. Thanks!

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@xtrawurst Look forward to your PR. Will close this ticket accordingly. Thanks!

closed 2019-12-05 by RealSenseCustomerSupport

shalkov, 2020-06-11:
Is this fix in the library RealSense?

Mad-Thanos, 2020-09-12:
> Is this fix in the library RealSense?

It seems this is not fixed yet.