realsenseai/librealsense #4215: librealsense on Android: "bulk_transfer on read endpoint returned error, ERROR: Out of memory"

xtrawurst, 2019-06-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1.100 | 
| Operating System & Version |   Android 8.0  |                    | 
| Platform                                 | Samsung Galaxy S8 (SM-G950F) |
| SDK Version                            |  2.22.0 | 
| Language                            |  C++ | 

### Issue Description
Configuring a pipeline with RGB 1280x720 and Depth 1280x720, then
```
rsPipeline->start(rsConfig);
rs2::frameset frames = rsPipeline->wait_for_frames(4000);
rs2::frame colorFrame = frames.first(RS2_STREAM_COLOR);
```
causes lots of
```
W librs   : bulk_transfer on read endpoint returned error, ERROR: Out of memory
```
on the Android log and no frames are captured.

xtrawurst, 2019-06-15:
This seems to happen more with higher resolutions and less with lower resolutions.

xtrawurst, 2019-06-15:
Update:
* Samsung Galaxy Phone S8 (SM-G950F): Happens all the time regardless of resolution. No successful frame transfer.
* Samsung Galaxy Tab S3 (SM-T820): Happens intermittently. Seems to happen more for higher resolutions. Usually frames are transmitted with some hickups when it happens.
* Samsung Galaxy Tab S4: Happens all the time regardless of resolution. No successful frame transfer.

GucciPrada, 2019-06-16:
Duplicate of #4091

xtrawurst, 2019-07-01:
@dorodnic @matkatz 
Update: Upgraded affected devices to Android 9.0, but the error still happens.

matkatz, 2019-07-04:
Hi @xtrawurst,
The Out of memory issue is not yet solved.
librealsense 2.24.0 improves some aspects of this issue but it seems that the phone can enter a state which only reboot can solve.
I will update once we have a progress with this issue.

xtrawurst, 2019-07-04:
@matkatz Thanks for the update.

If it helps, a while back we made a modified version of libRS 2.16.0 with a "hacked" libUSB that worked on non-rooted Android devices as a proof of concept internally in our company. That one had plenty of issues, but out-of-memory wasn't one. I.e. we were able to get an RGBD stream on devices that now show the out-of-memory error. I'd be happy to share the code if you're interested.

I understand that libRS's Android solution today is to do direct ioctl()s and not go through libUSB. If that's the case then there might be something that libUSB does differently that avoids the out-of-memory issue.

matkatz, 2019-07-09:
@xtrawurst sorry for the late response.
You are more than welcome to share your code or any suggestions.
I think that this issue appeared when I changed the backend from USB requests to USB bulk transfer.
The reason that I'm trying to avoid from going back is that on the devices that doesn't show this issue the performance is much better. i.e. I can stream HD depth + HD IR + 960x540 RGB all @ 30 FPS.
I'm currently trying to allow bulk transfer with smaller buffers in case I of OOM error.
It requires to assemble the frames from few packets and it still not stable enough.
In case that won't work I will check again the USB requests option.

xtrawurst, 2019-07-11:
@matkatz 
Ok, thanks for the explanation. Let's continue on #4091 then

matkatz, 2019-08-21:
duplication of #4091

closed 2019-08-21 by matkatz