realsenseai/librealsense #4216: Issues with Android wrapper, RS Application

maipbui, 2019-06-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Windows 10       | 
| Platform                                 | PC  |
| SDK Version                            |  2.0 | 
| Language                            |  C++  | 
| Segment			|  Smartphone (Samsung S8, S9)   | 

### Issue Description
Hello,

I installed 5 Android Applications to my phones, Samsung Galaxy S8 and Samsung Galaxy S9, just like this screenshot below:

[image]

When I access to "RS Camera" application, it always shows up the "Setting" with some features, it doesn't show anything else:

[image]

According to #3771 , I compared their images, then I figure I just have these features:

[image]

**And both my phones never show up the last two features:**

[image]

I did turn on developer mode and USB debugging mode. It keeps saying **Invalid configuration selected** on my phones. Is it because of my cables or my phones? Is is because I did not install those apps correctly? Is there any way to fix this issue?

Also, can you please drop me some information about how to use 5 apps of RS properly (RS Camera, RS Capture, RS Playback, RS Processing, RS Recording)?

Thank you!

matkatz, 2019-06-18:
Hi @maipbui 

The camera app currently doesn't prevent you from configuring illegal configuration.
Try to disable all streams and enable only depth-640X480-30 FPS.
If that doesn't work I will need some logs in order to understand what is the issue.

As for the 5 apps, I don't see any reason to use all of those.
All the apps except the "camera app" are sample code that demonstrate how to use specific features in the API.

maipbui, 2019-06-20:
Hi @matkatz 

Thanks for your response. It worked! 
The Camera App worked well when I tested it with different frame rates. However, the highest resolution it got is 640X480, is there any way to increase the resolution?

matkatz, 2019-06-20:
Please check issue #4091.
It seems that some devices enter to a state that limits the max possible USB transfers.
The issue is not resolved yet but for me and others rebooting the device eliminated the issue.

maipbui, 2019-06-21:
Yeah it seems that I have to reboot the device and set the lowest resolution at first, then increase it gradually.
I have same issue like yours also (unstable real time frame rate):

[image]
[image]
@matkatz Have you figured out it yet?
My goal is to have high resolution (1280X720) and stable real time frame rate (30FPS) for both RGB and depth streams when using this app. 
Last question: Does Intel Realsense @RealSenseCustomerSupport provide any Android devices that work well for this RS Camera? Or which Android devices do you recommend to use?

matkatz, 2019-06-30:
We don't have a recommended device currently.

I can say that with version 2.24.0 (also on the development branch) I'm able to stream depth 1280x720 @ ~27-30 FPS together with 960x540 @ ~27-30 over Pixel2, S8 and FireFly RK3399.

On both the Pixel and the S8 I have encountered the "Out of memory" issue which I didn't see on the RK 3399.

You can try to check the performance of the latest version using [the camera app.](https://play.google.com/store/apps/details?id=com.intel.realsense.camera&hl=en)

maipbui, 2019-06-30:
Sounds good. Thanks a lot for your answer.

closed 2019-06-30 by maipbui

ljxwy, 2019-07-28:
Excuse me, I want to ask a similar question. When I connect my phone with Realsense Camera and I access to "RS Camera" application, it always shows up "Failed to set streaming configuration",just like the screenshot below:
[image]

No matter what parameters(format/frame rate/resolution) I set , it displays the same interface above. I use a double type-c data cable to connect the realsense D435 and my phone(COL-AL10, Android 9). 
Is it because Android devices need root? How can I fix this issue?

 Thank you!

maipbui, 2019-07-29:
As mentioned above, "Try to disable all streams and enable only depth-640X480-30 FPS."

I think you should turn off developer mode &rarr; reboot the phone &rarr; turn on developer mode and usb debugging mode &rarr; disable all streams and enable only depth - with lowest resolution and lowest frame rate. (Also your phone should be fully charged)