realsenseai/librealsense #4217: Announcement: RealSense 400 Series firmware version 5.11.6.250 now available

MartyG-RealSense, 2019-06-15:
Hi everyone, 

A new firmware version **5.11.6.250** is now available.for RealSense 400 Series cameras.

https://downloadcenter.intel.com/download/28870/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

This firmware contains a fix for the **D435i** model - "Upgrade to latest production FW causes corrupted calibration table"

taletovic, 2019-06-17:
Hello, is there anything else besides FW update to do in order to fix the calibration / alignment issue
?

I've installed the new firmware and still getting the same problems using **RealSense Viewer v2.23.0** (latest version):
[image]

`rs-enumerate-devices -c` output:
[image]

And I am not able to calibrate the camera using **Dynamic Calibrator v2.6.8.0** (latest version):
[image]

zhuzunjie17, 2019-06-17:
Got the same result. All extrinsics corresponding to "color" are zero matrixs.

zhuzunjie17, 2019-06-17:
Can we use "Intel.Realsense.CustomRW -w -f xxx.xml" to write a good device calibration(provided by others) into our device? Then we can use Dynamic calibration to refine our calibration. If so, could anyone offer a device calibration file from "Intel.Realsense.CustomRW -r -f xxx.xml"?

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@taletovic @zhuzunjie17 Did you try "Intel.Realsense.CustomRW -g" to see if can reset the calibration to default gold factory setting?

taletovic, 2019-07-11:
@RealSenseCustomerSupport 
Hello, yes I've tried reseting the calibration to default gold factory settings but without any success.
Could you please provide us default calibration of a working D435i like @zhuzunjie17 said?

After that we could try to tune it again with Dynamic Calibration Tool.
Thanks

vhanded, 2019-07-13:
While calibrating using Dynamic Calibration Tool, I can complete the first step, where moving the target to clear the boxes. But second step scale calibration, it just remain 0%, no matter how I move the target.

RealSenseCustomerSupport, 2019-07-30:
----------------------------------------------

Hi @taletovic @zhuzunjie @vhanded,
There is another possible workaround that could recover RGB calibration table.

The recovery procedure is detailed in the firmware errata:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf
Please search for DSO-12586.

The firmware and the firmware update tool are located:
https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

Please let us know if the problem persists after applying the workaround. Thank you.

zhuzunjie17, 2019-08-11:
Sorry, I have exchanged the device, and the new device works well without the issue. 
However, I find that the solution of DSO-12586 mentions "Do not disconnect the device after updating the Firmware", so maybe it's not a perfect solution.

MartyG-RealSense, 2019-08-11:
Thanks so much for the update!

vhanded, 2019-08-11:
I performed a hardware reset, then run calibration again, the issue is gone.

closed 2019-08-20 by dorodnic