realsenseai/librealsense #4218: How to  configure RGB format

YH53, 2019-06-16:
Required Info
| Camera Model | { D415 } |
| Operating System & Version | {Win 10 |

| SDK Version | 2.0 |
| Language | C++ |
Platform: visual studio 2015

When I show color stream in this way:

 rs2::frameset data = pipe.wait_for_frames();
    rs2::frame color = data.get_color_frame();

    const int w = color.as<rs2::video_frame>().get_width();
    const int h = color.as<rs2::video_frame>().get_height();
    Mat image(Size(w, h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
    imshow(window_name, image);

 the color it shows might be wrong.blue things will turn yellow.

Where can I configure for color? How to configure RGB format for it? Thanks!

dorodnic, 2019-06-16:
Default format used by OpenCV is BGR
Default format used by librealsense is RGB
You can select format using `config.enable_stream`:
```cpp
config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH);
cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_BGR8);
pipeline p;
p.start(cfg);
```

YH53, 2019-06-17:
@dorodnic   Thanks，I've done it.

closed 2019-06-17 by YH53

akmaldeen, 2021-01-18:
I had the same issue, I have solved it with codes here. 

My complete codes:
 
int main() 
{
    config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH);
    cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_BGR8);
    pipeline p;
    p.start(cfg);

    while (true)
    {
        // Block program until frames arrive
        rs2::frameset frames = p.wait_for_frames();

        // Try to get a frame of a depth image
        rs2::video_frame depth = frames.get_color_frame();

        // Get the depth frame's dimensions
        const int w = depth.get_width();
        const int h = depth.get_height();
        Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

	imshow("window_name", image);
	waitKey(0);

    }

    
}