realsenseai/librealsense #4220: D

L1ttlewhite, 2019-06-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | v2.23 | 
| Operating System & Version | ubuntu 16.04   | 
| Kernel Version (Linux Only)    |          4.15.0-51-generic                        | 
| Platform                                 | PC |
| SDK Version                            |  lastest distribution                         | 

### Issue Description

My PC can not detect the realsense D435i, for the following erro when I use dmesg:
`[33417.878683] usb 2-2: USB disconnect, device number 10
[33500.986804] usb 2-2: new SuperSpeed USB device number 11 using xhci_hcd
[33501.011829] usb 2-2: New USB device found, idVendor=8086, idProduct=0b3a
[33501.011836] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[33501.011841] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435i
[33501.011845] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[33501.011849] usb 2-2: SerialNumber: 903223053620
[33501.045082] PKCS#7 signature not signed with a trusted key
[33501.053089] PKCS#7 signature not signed with a trusted key
[33501.057527] PKCS#7 signature not signed with a trusted key
[33501.059216] PKCS#7 signature not signed with a trusted key
[33501.060362] PKCS#7 signature not signed with a trusted key
[33501.063433] PKCS#7 signature not signed with a trusted key
[33524.766950] xhci_hcd 0000:00:14.0: Cannot set link state.
[33524.766963] usb usb2-port2: cannot disable (err = -32)
[33524.766974] usb 2-2: USB disconnect, device number 11
[33545.554970] usb 2-2: new SuperSpeed USB device number 12 using xhci_hcd
[33545.575974] usb 2-2: New USB device found, idVendor=8086, idProduct=0b3a
[33545.575981] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[33545.575986] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435i
[33545.575991] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[33545.575995] usb 2-2: SerialNumber: 903223053620
[33545.617983] PKCS#7 signature not signed with a trusted key
[33545.618502] PKCS#7 signature not signed with a trusted key
[33545.619374] PKCS#7 signature not signed with a trusted key
[33545.621121] PKCS#7 signature not signed with a trusted key
[33545.622317] PKCS#7 signature not signed with a trusted key
[33545.624051] PKCS#7 signature not signed with a trusted key
`

It can perform perfectly when I connect the T265 camera. I installed the distribution version of the librealsense. Last time, I installed the SDK by compling the source code and I could use the D435i. But it cannot perform well when I use the distribution version. Did I lost sth?

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@L1ttlewhite Seen from the dmesg you provided, D435i was detected. Aslo you can type "lsusb |grep 8086" to double check.  And could you please provide the result of "rs-enumerate"?

[33501.011845] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[33501.011849] usb 2-2: SerialNumber: 903223053620

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@L1ttlewhite Could you please update so that we can further support on this? Thanks!

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

@L1ttlewhite  Sorry that we didn't get response from you, will close the ticket at this point. Please feel free to open new ticket if you still have issue. Thanks!

closed 2019-07-22 by RealSenseCustomerSupport

chowjf, 2019-12-16:
@RealSenseCustomerSupport I have the exact same issue.
Here are some outputs:
1) lsusb | grep 8086
>Bus 001 Device 013: ID 8086:0b3a Intel Corp. 
2) rs-enumerate
>No device detected. Is it plugged in?