realsenseai/librealsense #4221: How to find orientation angle of an object using Intel Realsense D435i camera?

jaiatncson7, 2019-06-17:
Hi,

In my project, i need to find the distance and orientation angle of the object using D435i camera.
I can find the distance using Depth stream with depth_frame.get_distance(x, y) function. But i am not sure how to calculate the angle.

Can someone help me on finding the angle? 

Thanks for the help!

RealSenseCustomerSupport, 2019-07-24:
----------------------------------------------

Hi @jaiatncson7,

You like to find the attitude angle of target object in front of D435i camera? what type of object? a flat object?

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

I will close this if no further response.

closed 2019-12-05 by RealSenseCustomerSupport

vadirajkulkarni, 2020-12-04:
I am using D435 to detect the object and depth. I would like to find out the angle at which the object is placed.