realsenseai/librealsense #4222: Modify the resolution of viewer interface

MattixLee, 2019-06-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (RealSense SDK 2.0) | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC|
| SDK Version                            |  { 2.0}                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description
hello, interface of Intel RealSense Viewer gets bigger and I can't see the whole interface on the screen. What should I do to restore it to its original interface size?

Thank you !

dorodnic, 2019-06-17:
Can you share a screen-shot? Perhaps this is affected by global UI scaling set-up in Windows settings?

MattixLee, 2019-06-17:
[image]

dorodnic, 2019-06-17:
Thank you,
If you enter `%APPDATA%` + Enter in file Explorer, you should see file `realsense-config.json` in that folder. 
Try to delete it and restart the program. This will reset any persistent settings to defaults.

MattixLee, 2019-06-17:
Thank you for you help! It can work properly.

dorodnic, 2019-06-17:
Great! Sorry for the inconvenience. Most likely incorrect monitor dimensions were captured from one of the previous runs

closed 2019-06-17 by dorodnic

desert, 2020-07-07:
In my case this persists even deleting the file
[image]

desert, 2020-07-07:
[image]