realsenseai/librealsense #4223: Need distortion coeffs dumped in metadata csv from Realsense Viewer

kyranf, 2019-06-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | latest | 
| Operating System & Version |  Windows 10 | 
| Kernel Version (Linux Only)    |  N/A )                                         | 
| Platform                                 | X86 (x64) laptop |
| SDK Version                            | 2.22.0   | 
| Language                            | C++    | 
| Segment			|  Robots     | 

### Issue Description

Hello, 
I have captured some still depth-stream snapshots with Realsense Viewer v2.22.0 with my D415 and hoped to have enough useful information from the metadata .csv files generated during the snapshot. I hoped that I could load up the binary data and parse the .csv and have usable a dataset and could do the pixel deprojection. 

I have since found that the CSV doesn't dump the distortion coefficients after declaring what type of distortion model is used on the camera.  So I can build up an rs2 depth intrinisics struct up to the point of needing the coeffs. 

Can you guys please add that to the CSV dumps on the next version?

kyranf, 2019-06-17:
My plan is to make a Rectilinear image from the depth data and distortion model with the deprojection data to plot out the depth data in a X,Y rectified image for openCV, so I can do engineering-level analysis on the images.

ev-mp, 2019-06-17:
@kyranf , thank you for suggestion.
The metadata.csv was initially designed to include the video node metadata parameters, but there is no reason not to extend it to intrinsic params. It seem though, to be a corner use-case and therefore  will probably not be prioritized. If it is important to you then proposing a PR is an option you may consider.
 
In the meanwhile, running `rs-enumerate-devices -c > dev_capabilities.txt` will dump much of the capabilities including all extrinsic/intrinsic calibrations for off-line processing.

kyranf, 2019-06-17:
Thankyou for your response. 

For a previous vision guided robot application I developed I had the system which connects to the camera always dump the intrinics (just serialized the bytes of the struct) and stored to disc and a function to read from disc and deserialize and rebuild the intrinisics for offline processing/development. 

The human-readable nature of the metadata.csv was nice but as this project progresses, I will just take it into my own hands again.

I just thought i'd mention in case nobody had thought of it yet, as it would have been very handy for my early testing right now.

closed 2019-06-17 by kyranf

kyranf, 2019-06-17:
By the way, I just managed to get it to deproject pixel to point, without it even needing the distortion coefficients. Its with a D415 and Brown Conrady model. 

So I am able to do what I wanted, without the coefficients being saved to the metadata.csv :)