realsenseai/librealsense #4224: Aligned frames returned blank depth

filipematosinov, 2019-06-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Latest) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (4.18.0-21-generic)                                         | 
| Platform                                 | Up Board Squared  |
| SDK Version                            |  { 2.23.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  { }             | 

### Issue Description
I am having a issue with the alignment. I have recorded a bag file and when trying to replay it I get empty (i.e everything zeros) on the depth frame after aligning the frames (before is okay) while the color frame is always okay. Any clue why is this happening? 
Something interesting is that I am recording two cameras at the same time and only one has problems. Here is the code I am using to record them:

```
import pyrealsense2 as rs
import numpy as np
import cv2
import sys,os
from datetime import datetime
import time

if len(sys.argv) > 1: 
  record_time = int(sys.argv[1])
else:
  record_time = 30

path = 'Data/bag/'

if len(sys.argv) > 2: 
  if sys.argv[2] == 'local':
    local = True
  else:
    local = False
else:
  exit()

if len(sys.argv) > 3: 
  n_cams = int(sys.argv[3])
else:
  n_cams = 1

if os.path.isfile('rec_id.npy'):
  rec_id = np.load('rec_id.npy',allow_pickle = True) + 1
else:
  rec_id = 0

def start_record(cam_id = 1):

  pipeline = rs.pipeline()
  config = rs.config()

  path_name = path+str(rec_id)+"_cam_"+str(cam_id)+".bag"
  config.enable_stream(rs.stream.depth, 848,480, rs.format.z16, 15)
  config.enable_stream(rs.stream.color, 848,480, rs.format.rgb8, 15)
  config.enable_record_to_file(path_name)

  pipeline.start(config)

  return pipeline, path_name
  
pip, path_name = start_record(1)

if n_cams == 2:
  pip2, path_name2 = start_record(2)

print("Start recording")
time0 = time.time()

while (time.time() - time0) < record_time:
  continue

if n_cams == 2:
  pip.stop()
  pip2.stop()
else:
  pip.stop()

```

RealSenseCustomerSupport, 2019-07-19:
----------------------------------------------

@filipematosinov Did you have the issue if you only record and replay this issue camera? And for another good camera, did you also align frames?

filipematosinov, 2019-07-19:
I am not sure why but the problem is now solved. Both cameras seem to now align correctly. However when reading the bag file I do  

`pip = rs.pipeline()
    config = rs.config()
    rs.config.enable_device_from_file(config,
                                      read_path + filename,
                                      repeat_playback=False)

    config.enable_stream(rs.stream.depth, res_w, res_h, rs.format.z16, fps)
    config.enable_stream(rs.stream.color, res_w, res_h, rs.format.rgb8, fps)`

However when doing `pip.stop()` the script just stays there forever. What I am doing now is to not call pip.stop() but this unable the script to finished being forced to do control-c when I am sure it is done

RealSenseCustomerSupport, 2019-08-16:
----------------------------------------------

@filipematosinov Glad to see the issue resolved. Will close the ticket accordingly. Thanks!

closed 2019-08-16 by RealSenseCustomerSupport

filipematosinov, 2019-08-16:
@RealSenseCustomerSupport the issue is not solved. Can you please re-open it as you didn't give me any feedback?

redM0nk, 2020-03-16:
Hello. 

I'm noticing the similar issues when i align the depth frame to the color frame.
Camera is at height ~700mm and facing down at an empty conveyor.

Required Info |  
-- | --
Camera Model | { D415 }
Firmware Version | (5.12.3.00)
Operating System & Version | Linux (Ubuntu 16.04 LTS)
Kernel Version (Linux Only) | (4.15.0-88-generic)
Platform | (PC)
SDK Version | { 2.33.1 }
Language | {python}
Segment | { }
-- | --

RGB frame:
[image]

Corresponding Depth frame:
[image]
After aligning: The depth map above contains 0 values for all the pixels.
before aligning: the depth map contained showed good believable numbers.

Alignment is needed in my case since the depth map was a little behind the RGB color image (case when alignment was disabled).

Any feedback is appreciated.

tispratik, 2020-03-16:
Is "align"ment different than "synchronization" https://github.com/IntelRealSense/librealsense/issues/774 .... do we need synchronization?

tispratik, 2020-03-17:
@MartyG-RealSense can you please look into this? redM0nk is my colleague. We are facing issue with alignment of the frames as well. While trying to pick objects on moving conveyor, our robot is unable to go to the correct depth as the depth pixels are not aligned with the color pixels.

MartyG-RealSense, 2020-03-17:
@tispratik  As you are a Python user, a good starting point would be to check how you are implementing the alignment against Intel's Python tutorial for aligning depth and RGB streams into a single aligned image.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

tispratik, 2020-03-17:
Below is the python code that we've been using

```python

import pyrealsense2 as rs

align_to = rs.stream.color
align_handler = rs.align(align_to)

pipeline = rs.pipeline()
pipeline.start(config)

frames = pipeline.wait_for_frames()
frames = align_handler.process(frames)
color_frame = frames.get_color_frame()
depth_frame = frames.get_depth_frame()
depth_image_np = np.array(np.asanyarray(depth_frame.get_data())) if type(depth_frame) != np.ndarray else depth_frame
color_image_np = np.array(np.asanyarray(color_frame.get_data())) if type(color_frame) != np.ndarray else color_frame

MartyG-RealSense, 2020-03-17:
Python has another depth to color alignment example.   In that, they do the align operation after the pipeline has started and not before.  

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

redM0nk, 2020-03-17:
@MartyG-RealSense interesting. 
Perhaps the order is important. we are initializing the align process before starting the pipeline.
will change the order and try again.

redM0nk, 2020-03-17:
thanks @MartyG-RealSense 
After applying the above changes and powercycling the realsense device, it is now giving out good depth frame.

Another question, on the forum some people advice to apply the following changes on rgb sensor:

`sensor.set_option(rs.option.auto_exposure_priority, 0.0)`

Do you have any insights into what the above command does? and would you recommend using the above setting.

MartyG-RealSense, 2020-03-17:
It looks as though that instruction would disable Auto-Exposure Priority by setting it to '0'.  Disabling it, whilst having Auto-Exposure enabled, enforces a constant FPS speed instead of having the FPS able to vary depending on current environmental conditions.

redM0nk, 2020-03-17:
I'm using the camera in the following config:

```
RGB: 1280x720 @30 fps with exposure set to 32 ms
Depth: 1280x720 @30 fps with exposure set to Auto 
```
between frames, i'm doing some computation which roughly takes around 30 - 50ms. Sometimes i do notice frame drop, but it is mostly because the process is not consuming the frames fast enough.

Since rgb sensor is not set to auto-exposure, maybe i can disable the auto-exposure priority. what are your thoughts @MartyG-RealSense ?

MartyG-RealSense, 2020-03-17:
@redM0nk   Assuming the lighting is indoor and so controlled and stable (not changing much), I would be inclined to enable RGB auto-exposure to let the camera take care of the calculations.  Then experiment with turning Auto Exposure Priority on or off to see how it affects FPS.

tispratik, 2020-03-17:
@MartyG-RealSense we're using 32ms with additional lighting to fight motion blur for moving objects. Normally the auto exposure on color by default is 156ms... i would think that color stream is not slowing down when we are at 32ms. Though its very important for us to have depth exposure set to auto in order to get a good accuracy on depth data.

MartyG-RealSense, 2020-03-17:
@redM0nk  If you have strong lighting in the scene then it is possible to reduce exposure to around 1 ms, which also helps to avoid motion artifacts.  It sounds as though you have a manual exposure setup that works well for your application though.

redM0nk, 2020-03-18:
@MartyG-RealSense thanks for the help.
so after enabling the align_frame logic i'm noticing some frame drops. 
below is the snapshot from Datadog (where i'm logging the fps):
[image]

Anytime that i can do to optimize this?
I'm also looking into the following approach to help with the fps drop:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/frame_queue_example.py

MartyG-RealSense, 2020-03-18:
@redM0nk   It is possible to change the size of the frame queue to change the balance between performance and latency.  The more that it is weighted towards performance though, the higher the risk that there will be skipped frames.  

https://dev.intelrealsense.com/docs/frame-management#section-frame-drops-vs-latency

Bear in mind though that on less powerful hardware, running a logging / profiling program at the same time as your own program may in itself be a cause of lag as the logging consumes processing power.

redM0nk, 2020-03-18:
Makese sense.
@MartyG-RealSense Is it possible to store aligned frames (instead of regular frames) in the frame queue buffer. I looked for an example implementation but couldn't find one.

Regarding logging, yes that's true. We have a powerful hardware (i9 cpu) and logging is currently being done using a seperate background process (to avoid any unnecessary i/o)

MartyG-RealSense, 2020-03-18:
If you have a powerful computer with a lot of memory then an SDK feature called Keep() may give you better performance.  It stores frames in memory instead of writing it to storage.  When the pipeline is closed then you can then perform a batch processing operation such as post-processing or align on the stored frames and then save the frames all at the same time.

Because Keep() keeps the frames in memory though, the recording period may be about 30 seconds maximum before closing the pipeline, otherwise the computer would run out of memory.

The link below is an informative read.

https://github.com/IntelRealSense/librealsense/issues/1000