realsenseai/librealsense #4225: Python wrapper doc

ShettyHarapanahalli, 2019-06-17:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|745412070314|
|**Firmware Version**|05.10.03.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Documentation for the pyrealsense is  giving a error - 
WARNING: html_static_path entry '/home/suman/github/librealsense/wrappers/python/docs/build/_build/_static' does not exist 

I tried to build in the root, the html or pdf is generated in the path "github/librealsense/wrappers/python/docs/" 

Tried the make in the /doc/build it fail as well - 
WARNING: html_static_path entry '/home/suman/github/librealsense/wrappers/python/docs/build/_build/_static' does not exist

Can you please how to fix above error and steps to generate pdf version of document as well.

lramati, 2019-06-17:
The above is a warning and can be ignored.
The documentation is generated using Sphinx, it should just be a matter of
updating the arguments passed to sphinx

ShettyHarapanahalli, 2019-06-18:
Thanks for the response, Can I generate pdf document.

ShettyHarapanahalli, 2019-06-19:
Well html document is generated in the build folder, I did not look at it.  
librealsense/**build**/wrappers/python/docs/html

closed 2019-06-21 by ShettyHarapanahalli