realsenseai/librealsense #4226: Contradictory Camera Axis Labels for T265

cookem4, 2019-06-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715| 
| Operating System & Version |  Ubuntu 16| 
| Kernel Version (Linux Only)    |  4.4.38| 
| Platform                                 | NVIDIA Jetson|
| SDK Version                            |  2.0                           | 
| Language                            | N/A                          | 
| Segment			|  Robot         | 

I have been experiencing troubles with the labelled coordinate axis of the T265 camera as pictured below:
[image]

The problem I am having is that these axis do not match up the readings from the accelerometer in the realsense-viewer. The following photo shows the amount of acceleration on the Z axis when the Z axis is aligned with gravity:

[image]
The above image agrees with the labelled direction for Z, which the document also states is towards the back of the device.

The next image shows the amount of acceleration on the X axis when it is aligned with gravity:

[image]

This also agrees with the documentation which states that the X axis points towards the left side of the device where the RealSense logo is.

Now the problem I am having is with the Y axis. In order to align the Y axis with gravity, it must be in the opposite direction as labelled in the documentation. It is stated that the Y axis points towards the top of the device while the following photo shows that the Y axis actually points towards the bottom of the device:

[image]

My question is if this is a problem with the documentation on the camera axis or if it is a software related problem with displaying the acceleration readings? Based on the accelerometer measurements made to determine the axis, the coordinate system for the T265 is left handed. The coordinate axis on the accelerometer display shows that the coordinate system is right handed, which is due to the Z axis being mirrored from its true location.

ev-mp, 2019-06-18:
@cookem4 hello,
Thanks for reporting - this indeed seem to be documentation discrepancy.
The coordinate system you've mentioned applies to T265 output - the  **_Pose_** frame.
The IMU (raw) data does not follows this convention. We'll have this addressed in future releases.

cookem4, 2019-06-18:
@ev-mp 
Thank you for the prompt reply,

After looking into the issue further with the pose display on the RealSense-Viewer application, there seems to be some discrepancy between the pose of the camera, the fixed frame, and the quaternion that is reported.

The following photo shows the quaternion reported when the axis of the camera's pose and the fixed frame are aligned:

[image]

Reported as a quaternion in [x y z w], this has values of [0 -1/root(2), 1/root(2), 0] which corresponds to the following rotation:

[image]

This rotation maps X to -X, Y to -Z, and Z to -Y, which does not match at all with what is shown in RealSense-Viewer. Since the axis are perfectly aligned in RealSense-Viewer, the quaternion should be [0 0 0 1] but that is not the case.

Do you know why the quaternion is reported this way and what is actually happening with the reported pose information?

dorian3d, 2019-06-18:
Hi @cookem4. The issue with the accelerometer values is definitively a lack in the documentation. The coordinate system you see in the doc only applies to the T265 pose stream; gyro and accelerometer (IMU) have a different coordinate system. 
I drew them [in this image](https://github.com/dorian3d/librealsense/blob/t265-doc/doc/img/T265_sensor_extrinsics.png). It should be consistent with your observations.

With regard to the quaternion, I'd say it is a visualization issue. T265 poses are relative to an `origin` coordinate system which, as far as I know, is not explained in the documentation (we can see it as in [this image](https://github.com/dorian3d/librealsense/blob/t265-doc/doc/img/T265_orientation_axis.png)). When tracking starts, the `origin` is created in the current T265 position, with Y axis aligned with gravity and pointing to the sky always. Origin's X and Z axes depend on the initial orientation of the device, and they form a right-handed system. It is deterministic, though, so if you start tracking with T265 at the same position and orientation, your `origin` will be defined the same always. In the specific case in which you start tracking with the fisheye sensor looking forward, `origin` will be defined the same as the T265 pose stream (X pointing right, , Y up, Z backward).

Coming back to the quaternion, given the explanation about `origin` above, If I just start tracking in RealSense Viewer with the camera looking forward, I get `translation [0 0 0]` and `rotation [0 0 0 1]`, as expected, meaning the current T265 pose is exactly the same as the origin. However, I see the Viewer shows two coordinate systems with X pointing in opposite directions. To make the rendered coordinate systems be the same, I have to rotate the camera 180 deg around Y and 90 deg around X, getting the quaternion you see. So, my conclusion is `origin` is not correctly displayed in Viewer.

ev-mp, 2019-08-05:
The documentation have been updated since v2.24 release

closed 2019-08-05 by ev-mp

lms0, 2020-09-11:
> 
> **Update**: we updated the [documentation in `development` branch](https://github.com/IntelRealSense/librealsense/blob/development/doc/t265.md).
> 
> With regard to the quaternion, I'd say it is a visualization issue. T265 poses are relative to an `origin` coordinate system which, as far as I know, is not explained in the documentation (we can see it as in [this image](https://github.com/dorian3d/librealsense/blob/t265-doc/doc/img/T265_orientation_axis.png)). When tracking starts, the `origin` is created in the current T265 position, with Y axis aligned with gravity and pointing to the sky always. Origin's X and Z axes depend on the initial orientation of the device, and they form a right-handed system. It is deterministic, though, so if you start tracking with T265 at the same position and orientation, your `origin` will be defined the same always. In the specific case in which you start tracking with the fisheye sensor looking forward, `origin` will be defined the same as the T265 pose stream (X pointing right, Y up, Z backward).
> 

If I understand correctly what you mention, the pose data coordinate system is such that:

- +Y points to the sky

- -Z is the projection from the optical axis **from top to ground**

Then the initial rotation around y should be always 0 degrees, because Y axis is fixed, and a rotation around the Y axis would lead to a mismatch for the second condition. Right?