realsenseai/librealsense #4229: T265  connction error

ttomchy, 2019-06-18:
Hello, I am trying to connect realsense d415 and t265 at the same time on my computer. However   when I try to use  ```realsense-viewer```  to see whether the cameras works or not, sometimes I couldnot found t265 device. When I trying to unplug t265 and connected it again, then ```realsense-viewer``` can found the device. Did anyone meet similar problem before. I am confused ...

mickeyouyou, 2019-06-25:
I also encountered an unstable connection in the same environment. Sometimes the device can be seen through the realsense-viewer, sometimes not at all. An exception was encountered while reading with the sample rs-pose-and-image: "No device connected".

RealSenseCustomerSupport, 2019-07-24:
----------------------------------------------

Hi @ttomchy, @mickeyouyou,

This should be similar with #4331.
1. What's your OS? Is T265 recognized as USB2 when you connect T265 without running any librealsense app like realsense-viewer or rs-pose?
2. Are both d415 and t265 on one USB3 hub and then connected to the host?
3. Please also show "lsusb" and "dmesg" log before and after you run the librealsense app.

RealSenseCustomerSupport, 2019-08-15:
----------------------------------------------

Hi @ttomchy, @mickeyouyou,

Any update? Is the issue clarified?

mickeyouyou, 2019-08-15:
maybe this is my codes problem. It has been sovled. thank you .

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

Hi @ttomchy,

Do you also clarify the issue?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @ttomchy,

Any update? Is any USB hub used for T265 and D415?

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

I will close this if no further question.

closed 2019-12-06 by RealSenseCustomerSupport