realsenseai/librealsense #4231: Temporal Filter - Persistency Mode

Bedrettin-Cetinkaya, 2019-06-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.21.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  others                           | 

### Issue Description

I used temporal filter with option "Persist Indefinitely". However, i need to clear this persistency sometimes. Is there any way to clear this persistency without restarting camera?

Thanks.

dorodnic, 2019-06-18:
Hi @Bedrettin-Cetinkaya 
I'm semi-guessing, but `temp = temporal_filter()` would most likely do the trick (persistence is saved at the processing block object level, so emptying it should do the trick). @ev-mp do you know?

Bedrettin-Cetinkaya, 2019-06-18:
deleting filter solved the problem.

closed 2019-06-18 by Bedrettin-Cetinkaya