realsenseai/librealsense #4232: How can i create a ROI in MATLAB

Deresix, 2019-06-18:
----------------------------------------------------------------------------------------------------

| Required Info                         | Create a ROI in MATLAB                         |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 10 |                                   
| Platform                                  | PC  |
| SDK Version                            |  2.0                                 | 
| Language                                | MATLAB                         | 

### Issue Description
Hi everybody, I'm quite a newbie in both MATLAB and this community.
I managed to acquire an object as a pointcloud, but now I need to limit the region of interest the camera has to deal with. Can anybody help me with the code?

I started editing functions of the roi_sensor class but can't find out how to make this class interact with the matrix or even with the pointcloud object.

Thanks in advance for your help.

MartyG-RealSense, 2019-06-18:
I hope that the link below will be of help to you.

https://forums.intel.com/s/question/0D50P00004EW22rSAD/how-do-you-select-the-roi-in-depth-frame?language=en_US

Deresix, 2019-06-19:
Thanks for your response, unfortunately I already had a look at it but I think I need more examples of MATLAB application with this hardware. I decided to try to export the obtained pointcloud in a standard .ply file. That could be possible (as far as i understood) using the save_to_ply class.
However when i type `save_to_ply('example.ply',pointcloud)` it returns me the error 

> The specified superclass 'realsense.processing_block' contains a parse error, cannot be found on MATLAB's search path, or is shadowed by another file with the same name.

How can i fix it? I tried adding all the folders to matlab path but nothing changed..

The save_to_ply class is

 `% Wraps librealsense2 save_to_ply class

classdef save_to_ply < realsense.processing_block
    properties (Constant=true)
        option_ignore_color = realsense.option.count + 1;
    end
    methods
        % Constructor
        function this = save_to_ply(filename, pc)
            switch nargin
                case 0
                    out = realsense.librealsense_mex('rs2::save_to_ply', 'new');
                case 1
                    validateattributes(filename, {'char', 'string'}, {'scalartext'});
                    out = realsense.librealsense_mex('rs2::save_to_ply', 'new', filename);
                case 2
                    validateattributes(filename, {'char', 'string'}, {'scalartext'});
                    validateattributes(pc, {'realsense.pointcloud'}, {'scalar'});
                    out = realsense.librealsense_mex('rs2::save_to_ply', 'new', filename, pc.objectHandle);
            end
            this = this(at)realsense.processing_block(out);
        end
        
        % Destructor (uses base class destructor)
    end
end

MartyG-RealSense, 2019-06-20:
I will tag one of the Intel team on this forum to take a look at your question, as it is outside of my knowledge.    @dorodnic

lramati, 2019-06-20:
Replacing line 1 with `classdef save_to_ply < realsense.filter` and line #20 with `this = this@realsense.filter(out);` should fix your problem.

Deresix, 2019-06-20:
Thank you so much! It worked perfectly. 
To anybody who'd like to manage ROIs using the realsense matlab wrapper I'd suggest to transform the vertices matrix obtained with the pointclud_example into a pointCloud file. Then you can use the proper class which has much more documentation and tutorials.

Again, thanks for your patience and help.

closed 2019-06-20 by Deresix