realsenseai/librealsense #4233: Segmentation Fault on rs2_frame_add_ref function

felipeccc, 2019-06-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   Linux Ubuntu 16.04.06  | 
| Kernel Version (Linux Only)    |  (4.15.0-51-generic) | 
| Platform                                 | PC |
| SDK Version                            |  { Intel RealSense SDK 2.0 (2.23.0) }  | 
| Language                            |  C++  | 
| Segment			|            others                |

### Issue Description
I built the library from sources, but when I tried to run a program that uses the library, I get the following output with gdb:

> Thread 24 "XXXXX" received signal SIGSEGV, Segmentation fault.

> 0x00007fffbaea3692 in **rs2_frame_add_ref** (
        frame=0x7fffbbecfdc8` <vtable for librealsense::depth_frame+16>, 
        error=0x7fffcb7fd3a0) at /home/detect/librealsense/src/rs.cpp:856
>856	    ((frame_interface*)frame)->acquire();

I checked the function on rs.cpp and since it only calls the acquire() function when the frame is not null, I do not understand where the error could be :

`void rs2_frame_add_ref(rs2_frame* frame, rs2_error** error) BEGIN_API_CALL`
`{`
    `VALIDATE_NOT_NULL(frame);`
    `((frame_interface*)frame)->acquire();`
`}`

How could I try to find where does the error comes from ?

dorodnic, 2019-06-18:
Well, this really depends on what you pass as the value of `rs2_frame* frame`. Most likely it is an invalid pointer (already de-allocated).

felipeccc, 2019-06-19:
Thank you. You are right, when I changed the way I was compiling the other library, it worked.
I'm closing this issue.

closed 2019-06-19 by felipeccc

mbechard, 2019-06-25:
@felipeccc What did you need to change to fix the crash? I'm also seeing the same crash even though the frame I'm getting is directly out of a rs2::frameset

felipeccc, 2019-06-26:
@mbechard I don't know if it will be of much help, but I am using a home made library (that uses realsense2) and the problem was solved when I built it in the Release Mode instead of Debug.
And even though it crashed in a realsense function, I still don't know the source of the problem.