realsenseai/librealsense #4234: Using Intel D415/D435 in humid environments

amaanda, 2019-06-18:
Hello

For my application, it would be possible that the camera might be exposed to humid/cold weather. I understand since it is a camera this kind of condition should be avoided, I am only studying the options we have.
On the D400 series datasheet, there is information about storage but not about operating in humid weather:
[image]

Me and my team are currently working with a stereo camera which, among our units, one got foggy lenses due to humidity penetrating the camera after sending it to a cold/humid weather place. We are producing a case for the camera; but we are also considering switch to Intel cameras, so my biggest worry is those holes they have along the case.

I wonder if the Intel team have done any test under these conditions and if possible, please share.

MartyG-RealSense, 2019-06-18:
There was a recent discussion about RealSense and humidity conditions on the Intel Support forum.

https://forums.intel.com/s/question/0D50P00004LfpD2SAJ/can-the-d435-work-in-a-humid-environment?language=en_US

MartyG-RealSense, 2019-06-18:
RealSense cameras can operate in wet conditions such as underwater but they and their lenses are not waterproof.  If you are able to produce a custom case, the best solution may be to use a RealSense 'Depth Module Kit', a version of the camera that has no casing, so that you can fit it into your own casing design to keep it waterproof.

If you would like to go the self-assembly route with RealSense, the link below may be useful to you.

https://forums.intel.com/s/question/0D70P000006JeBpSAK

amaanda, 2019-06-19:
@MartyG-RealSense can you recommend any robust case / container supplier?

MartyG-RealSense, 2019-06-19:
Most people who need a custom casing or mounting solution 3D-print it (using either their own design or a model blueprint downloaded from the internet).  So options for pre-made ones purchasable from a supplier are slim.  MobileDemand offer a rugged protective boot for the D415 cased USB camera.

https://www.ruggedtabletpc.com/blog/3d-camera-for-rugged-tablet

[image]

closed 2019-07-11 by RealSenseCustomerSupport