realsenseai/librealsense #4235: Error when trying to use matlab wrapper

sharfin99, 2019-06-18:
I am at the point where I am supposed to build librealsense.mex, but I keep getting several errors. First, I kept getting errors saying visual studios was missing libraries and headers. I just copied those into the same directory and that fixed the problem. Now, im getting an error that says: "cannot open file realsense2.lib". Did I configure cmake wrong or is there another problem. I am extremely desperate at this point and have been trying to get this matlab wrapper to work for 11 hours now. Please help!

dorodnic, 2019-06-18:
Hi @sharfin99 
Sorry for the frustration, I'd like to make sure you have tried / considered the [Windows installer](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#windows-installer) route?

sharfin99, 2019-06-18:
@dorodnic I am very new at all of this, so I did not know that was a different option than the "buid from source" option. I did select that option when downloading the sdk, but matlab still does not recognize objects and functions from the realsense camera. Do I have to move files or is it something else that I am missing?

dorodnic, 2019-06-18:
From my personal experience setting-up the Installer version is just simpler. 
You need the `+realsense` folder to be near your Matlab project, so it detect the declarations.
Inside you will also find couple of examples, such as `depth_example.m`
@lramati please comment if anything extra is required

lramati, 2019-06-20:
Nope. As step 3 of the Installer options states, you just need to make sure Matlab can find the files. Your options are either to add it to Matlab's path (this can be done from within the matlab's settings, and there should be plenty of tutorials on the internet showing how), Or simply copying `+realsense` (The **Whole** folder, not just it's contents) into your project's root folder

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@sharfin99 Did you get through the issue now with Dorodnic and Iramati's suggestion? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

@sharfin9 Could you please update? Thanks!

sharfin99, 2019-07-18:
Fixed! I did not realize the wrapper and download are the same thing. I just needed to add the path to the folder in MATLAB.

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

@sharfin99 Glad to hear that it fixed. Close the issue accordingly. Thanks!

closed 2019-07-22 by RealSenseCustomerSupport