realsenseai/librealsense #4236: Intel RealSense Sample Issues with MVS

wanninng, 2019-06-18:
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win (10) | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            | {python} | 
| Segment			|  {Robot/Smartphone/VR/AR/others }  | 

### Issue Description
I'm a total beginner to this essentially, I'm currently trying to open up the samples with Microsoft Visual Studio to check them out. I was wondering if there was a guide to all of this, I've spent a bit of time on GitHub reading around downloading things, following installation instructions and what not - to end up with no luck. Anyways currently I can't open up any of my samples due to an error. 
[image]

(Note - Tried to use CMake but don't know how, always gave me a error)

Samvid95, 2019-06-21:
Hey are you sure you are doing this correctly? 

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_windows.md

These are fairly straight forward steps with screen shots.

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@wanninng Did you get it through? Please refer to https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_windows.md as Samvid95's replied. Looking forward to your update. Thanks!

Samvid95, 2019-07-17:
Nothing is updating in this thread. I think we should close this thread.

wanninng, 2019-07-18:
Sorry about that, I didn't realize that I received notifications from this thread. I tried @Samvid95 link after uninstalling everything and re-downloaded everything. It finally worked and I thank you guys for your help!

closed 2019-07-18 by wanninng