realsenseai/librealsense #4238: Add configuration option to call CoInitializeEx with COINIT_APARTMENTTHREADED flag

tim-depthkit, 2019-06-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   | N/A | 
| Operating System & Version |  Win (8.1/10) | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC  |
| SDK Version                            | 2.20.0 and up | 
| Language                            |  C  | 
| Segment			|  VR/AR | 

### Issue Description
Currently the WMF backend initialized COM as follows:
`CoInitializeEx(nullptr, COINIT_MULTITHREADED); // when using COINIT_APARTMENTTHREADED, calling _pISensor->SetEventSink(NULL) to stop sensor can take several seconds`

However in applications that have COM initializations on other threads using COINIT_APARTMENTTHREADED (like the Qt framework, for example), this can lead to a deadlock inside COM functions.

As the comment suggests, there is a reason for initializing in this way, but the SDK user should be allowed to configure this if necessary, via a CMake build option or similar.

dorodnic, 2019-06-19:
Yep, that make sense. Do you think this can be covered as a CMake parameter or dedicated API?

tim-depthkit, 2019-06-19:
A CMake option would be sufficient for my use case.

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

We are planning on implementing this. The rough schedule is Q3. Thanks.

RealSenseSupport, 2020-11-12:
The change has been merged into the development branch. Please let us know if there is any concerns. 
Thank you.

RealSenseSupport, 2020-11-18:
This enhancement is included in the v2.40.0 release. Thank you.
[Release Notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400)  and search for DSO-15197
Closing this case.

closed 2020-11-18 by RealSenseSupport

kayychou, 2024-01-24:
Hi @tim-depthkit, @dorodnic,

I encounter issues using the Qt framework after initializing my Intel camera D415 so reading these posts is really helpful to understand the root cause.

My issue (Main-UI blocked when opening a QFileDialog) seems to be solved by using librealsense compiled with the cmake option COM_MULTITHREADED set to OFF.

Could you please give me more insights regarding the impact this option can have on the SDK behaviour ?
The doc mentions that "stop sensor can take several seconds".
@tim-depthkit Have you noticed unwanted side effects with the option COM_MULTITHREADED set to OFF ?
@RealSenseCustomerSupport to what extent do you recommend to preferably use the SDK with this option set to ON ?

Thanks for your support!

MartyG-RealSense, 2024-01-24:
Hi @kayychou  The only available RealSense related reference to COM_MULTITHREADED is a line in the **lrs_options** CMake file that provides for COM_MULTITHREADED the description "Set OFF to initialize COM library with COINIT_APARTMENTTHREADED".  It is set to On by default.

https://github.com/IntelRealSense/librealsense/blob/master/CMake/lrs_options.cmake#L8C27-L8C90

A Microsoft reference at the link below suggests that COINT_APARTMENTTHREADED should be used (i.e set to On) when there is a thread that creates a window.  Otherwise, do not use COINT_APARTMENTTHREADED.

https://learn.microsoft.com/en-us/windows/win32/learnwin32/initializing-the-com-library