realsenseai/librealsense #4239: Compiling for Raspberry Pi 3B+ from Windows?

reyes73a, 2019-06-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                
|---------------------------------|------------------------------------------- 
| Camera Model                       | { D435, T265 } 
| Firmware Version                   | librealsense ver: 2.23.0 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 16/18)                                  
| Platform                                 | PC/Raspberry Pi|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          
| Language                            |  {C/C++ }                          
| Segment			|  {Robot/others }                          

Hi guys,

I am currently deploying the librealsense library on a Raspberry Pi 3B+. I have tried both D435i and T265 and they both seem to work. My current procedure has been to compile from the Pi board directly. The Pi board is currently running **ubuntu-mate-18.04.2-beta1-desktop-armhf+raspi-ext4**. When compiling from the Pi board directly it takes forever (several hours) - not to mention that it requires a swap file due to RAM issues during building process - and I would like to find a faster alternative to this. Is it possible to compile for Pi 3 from Windows or Ubuntu 16.04 and then move the files over to the Pi 3? If so, how can I do this? Keep in mind that I am new to Visual Studio, C/C++ and its compilation process ( I typically work on Java). Any help on this issue would be greatly appreciated!!!!

Thanks a lot!

Dan.

RealSenseCustomerSupport, 2019-07-19:
----------------------------------------------

@reyes73a You can search the cross building toolchains which might help you.

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

@reyes73a Any other questions about this? Thanks!

closed 2019-08-01 by RealSenseCustomerSupport