realsenseai/librealsense #4241: object space coordinate

buctfy, 2019-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I want to know the space coordinates (x,y,z)of  a point in rgb frame. 
My ideal is  
first  got point's coordinate in rgb frame, then align this point to depth frame,finally get point's space coordinate. But according to Camera calibration,a point in depth can transform to rgb frame in pixel coordinate,can't do it  reverse,because rgb frame lack distance information (z). I know rs300 can get face space coordinate  in real time and want to know  the principle of it

ev-mp, 2019-06-19:
@buctfy , the SDK provides `rs2_project_color_pixel_to_depth_pixel` functionality that addresses this use-case.
Using this API you'll find the depth pixel corresponding to the RGB pixel you're interested in, then use `rs2_deproject_pixel_to_point` to convert depth 2D pixel to X,Y,Z coordinate.

buctfy, 2019-06-21:
> @buctfy , the SDK provides `rs2_project_color_pixel_to_depth_pixel` functionality that addresses this use-case.
> Using this API you'll find the depth pixel corresponding to the RGB pixel you're interested in, then use `rs2_deproject_pixel_to_point` to convert depth 2D pixel to X,Y,Z coordinate.

It is strange that the function rs2_project_color_pixel_to_depth_pixel can't  be found in https://intelrealsense.github.io/librealsense/doxygen/annotated.html.
I want to know the principle of  function rs2_project_color_pixel_to_depth_pixel. I saw the code from https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h  ,but I can't understand the code from135 to215,could you please explain them to me?

ev-mp, 2019-06-23:
In general it is not possible to project RGB (or any other 2D stream) to the depth stream directly, as you've rightfully noted, and projecting the whole depth frame to RGB to find a single pixel is not an option.
Luckily, we utilize the stereo vision feature called [`epipolar constrains`](https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md#calibration-and-rectification) which intuitively means that a single pixel's projection (a ray "fired") onto 3D space can be intersected only by a separate sensor's pixel that belong to the same `epipolar plane`.
This greatly reduces the complexity and limits the search range to be along a single line of depth pixels.

With this premise in place, the devised algorithm runs as follows:
**Initialization**:
- Guess two depth values between which the depth resides. In practise there is no need to check below ~25cm and more than TBD meters (establsh min/max boundary).
-  Transform the min/max RGB points to the Depth frame to find the two corresponding depth pixels. Make sure that the depth pixels are within the depth frame boundaries and crop to border if necessary.
- The two obtained depth pixels (X0,Y0, X1,Y1) define a line segment along which the match will be looked up.  

**Search**:
- Traverse along the line of depth pixels between (X0,Y0)  and (X1,Y1)
- For each depth pixel (D_i) in range:
   - Calculate the D_i's correspondence in the RGB frame (R_i). I.e Deproject->Transpose->Project
  -  Measure the 2D distance between the R_i and the original RGB pixel.
  - If R_i is closer to the target RGB pixel than the previous attempts - map this pixel as a candidate and store the depth pixel that originated this match.
 - When finished, the best candidate will be promoted to a match and returned to the user.

HIAI2019, 2019-07-13:
> In general it is not possible to project RGB (or any other 2D stream) to the depth stream directly, as you've rightfully noted, and projecting the whole depth frame to RGB to find a single pixel is [...]

Thanks a lot.But I can't understand the argument of function next_pixel_in_line ,I think the coordinates of  pixel should be int not float.
[image]

ev-mp, 2019-08-20:
The values are trimmed to the depth pixels grid by the caller code. It is clear that the algorithm could be further optimized, but so far we're satisfied with the results.

Do you need more assistance on this ?

closed 2019-09-03 by HIAI2019