realsenseai/librealsense #4242: Nodejs no device connected though it works through the intel realsense viewer

richieclark33, 2019-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   windows 10.0.17134  |                                          | 
| Platform                                 | PC- surface book 2 |
| SDK Version                            |  Latest - I tried Build 2.23.0, but went back to 2.21  (however because it's node I'm using the node-librealsense npm for the actual package)                         | 
| Language                            |  nodejs                     | 
| Segment			|                     | 

### Issue Description
I have been able to get the RealSense Viewer 2.21 to recognize my camera and it works great.  If I run examples in C++ the camera seems to recognize as well.  I was able to use 

`npm install --save node-librealsense`

and get the package to output and build the files.

When I try and run the nodejs examples, I get a rs2_pipeline_start : No device connected error.  I get that error for all the examples in the examples folder as well as just the sampel usage demo code on the package page.  

Since I was able to get my camera to recognize through the viewer, I grabbed my serial number from there and then tried to do this:
`var config = new rs2.Config();
config.enableDevice("1111myserialnumber");
const pipeline = new rs2.Pipeline();  // Main work pipeline of RealSense camera
pipeline.start(config);  // Start camera`

I then got the same error but through the rs2_pipeline_start_with_config function instead of the 
of just the rs2_pipeline_start

I tried through C# as well and was having similar issues.  I've uninstalled, reinstalled, tried a bunch of different versions etc but I'm having a hard time finding it where people recognized the camera in the realsense viewer but then couldn't in the wrappers.  I also tried to trace a bit through the actual code in the librealsense, but it's just a bit complicated to go through when not knowing exactly what might cause the device not to be found.

I was hoping someone could suggest some other troubleshooting options or ideas to try and track down why the nodejs wrapper can't find my device?

dorodnic, 2019-06-19:
npm package is somewhat outdated, but if you build the wrapper from source it will work

richieclark33, 2019-06-19:
When I build the source, am I supposed to be following 1.2 or 1.3 on this page : 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs

I tried doing 1.2 which is this page : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

and checked BUILD_NODEJS_BINDINGS, but I'm not sure once I do that what I'm supposed to take from that in order to reference in node as the binaries that it builds don't seem to be what's needed and I don't see any js files exported.

this was the log on the build in CMake - https://www.screencast.com/t/blYiZUT3 
maybe it didn't actually build anything and it should have more output?

when I follow 1.3 doing this command on the main directory
`cmake -DBUILD_NODEJS_BINDINGS=1`

 I get this output : https://www.screencast.com/t/taY6AcpmW
 I go to wrappers/nodejs/examples run npm install and then run one of the examples and I get a
"cannot find module 'bindings'" error which I'm assuming means the cmake command didn't execute properly?

RealSenseCustomerSupport, 2019-08-27:
----------------------------------------------

When you are building wrapper from source you should follow all the instructions from "Build from Source" section.That means that paragraphs 1.2 and 1.3 should be both fully completed.

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

Hi richieclark33,

Do you need any further clarifications regarding this topic?

closed 2019-09-07 by richieclark33