realsenseai/librealsense #4243: Can't find device

pangchenglin, 2019-06-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Operating System & Version | Linux Ubuntu 16.04  | 

### Issue Description
I can open the realsense-viewer, but it can't find my camera.
I try update the firmware. But it still can't connect my camera.
[image]
When i try
 `lsusb`
[image]

When i try
 `intel-realsense-dfu –b 002 –d 002 –f –i /home/lilala/Signed_Image_UVC_5_11_1_100.bin`
[image]
When i try
`intel-realsense-dfu –p`
[image]

How can i use the camera?

ev-mp, 2019-06-19:
@pangchenglin,
You have the device enumerated in USB2 mode: 
> lsusb
...
8086:**0ad6** Intel Corp.

D435i was initially released with USB3 support, and with FW versions prior to v5.11.6.200 it will not run in USB2 mode.

If that was not the intend then just check the cable/switch port to make sure it is recognized as USB3 (8086:0b3a).
In case you have to use it with USB2 connection then upgrading to the latest recommended v5.11.6.250 is required.

pangchenglin, 2019-06-20:
I change another computer, it can work. But when I try to install realsense-ros, I have a new problem. I submit the problem in https://github.com/IntelRealSense/realsense-ros/issues/821 .
How can I solve it?

closed 2019-07-09 by RealSenseCustomerSupport