realsenseai/librealsense #4244: get from color IR left right frame from D410

kazu0622, 2019-06-19:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410                                                      | 
| Firmware Version                   |05.10.06.00                                            | 
| Operating System & Version |  Win 10                                                | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  {  2.<20>.<0> }                          | 
| Language                            |  {C}                          | 
| Segment			|  others                           | 

### Issue Description
Hello.

I'm trying to get colored-IR left and right frame data from D410, but ,failed.
My description is as below when I failed.
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8);
cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8);
rs2::pipeline_profile selection = pipe.start(cfg);
rs2::frameset frameset = pipe.wait_for_frames();
auto ir_frame_left = frameset.get_infrared_frame(1);
auto ir_frame_right = frameset.get_infrared_frame(2);
This description worked for D435 but D410.

On the othrer hand, It worked for D410 as below.
rs2::pipeline pipe;
rs2::pipeline_profile	 selection = pipe.start();
auto ir_colored_left = frameset.get_color_frame();

I'd like to know how to get both ir_colored_left and right frames from D410, if you would.

Thanks.

dorodnic, 2019-06-19:
The hardware currently can either output single RGB infrared stream or two grayscale infrared streams, but not double RGB infrared streams or RGB left + grayscale right.

ev-mp, 2019-06-19:
@kazu0622, the syntax for Left IR as RGB is -
>cfg.enable_stream(RS2_STREAM_INFRARED, ~~1~~**0**, 1280, 720, RS2_FORMAT_~~Y8~~**RGB8/BGR8/..**);

kazu0622, 2019-06-20:
@dorodnic,ev-mp. Thank you for your reply.
Could you tell me how to  describe the program that I can get both grayscale left and right?
The description below doesn't work.
 rs2::config cfg;
 cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8);
 cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8);
 rs2::pipeline_profile selection = pipe.start(cfg);
rs2::frameset frameset = pipe.wait_for_frames();
 auto ir_frame_left = frameset.get_infrared_frame(1);
 auto ir_frame_right = frameset.get_infrared_frame(2);

Thanks.

kazu0622, 2019-06-20:
@dorodnic,ev-mp Thanks to your advice, I've done.
I was able to solve the probrem just focusing to get Depth, ir-gray-left, ir-gray-right.
I really appreciate to your advice.
Thank you.

closed 2019-06-22 by dorodnic