realsenseai/librealsense #4245: How to align depth and rgb frame in matlab

Ahad19931, 2019-06-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (5.11.06.200) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.22.0 }                          | 
| Language                            |  {MATLAB }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi,
Can someone please guide me how to align the depth frame with the RGB frame in MATLAB? I want to perform the similar task as is given in the rs-align-advanced example where applying threshold to the depth value also clip the rgb frame accordingly. For example, if the threshold is set to 1m, the rgb frame will be clipped to 1m. There is no such example present in MATLAB so I would be pleased if any can help me performing the same task in MATLAB.

lramati, 2019-06-19:
The Matlab API mimics the C++ API, so the core logic should be easily
portable

Ahad19931, 2019-06-19:
Thanks for the reply, Ok I will try and let you know if the problem still exists.

Ahad19931, 2019-06-19:
I tried to align the depth and the rgb frame but I not sure it worked because the two frames are still not align. Below you can see the output frame I got using the following code:
% Make Pipeline object to manage streaming
  pipe = realsense.pipeline();
  % Start streaming on an arbitrary camera with default settings
  profile = pipe.start();
  align_to = realsense.stream.color;
  alignedFs = realsense.align(align_to);
  % Get streaming device's name
  dev = profile.get_device();
  name = dev.get_info(realsense.camera_info.name);
  % Get frames. We discard the first couple to allow
  % the camera time to settle
%while(true)
  for i = 1:5
    fs = pipe.wait_for_frames();
  end
  % Select depth frame
  depth = fs.get_depth_frame();
  % get depth image parameters
  depthSensor = dev.first('depth_sensor');
  depthScale = depthSensor.get_depth_scale();
  depthWidth = depth.get_width();
  depthHeight = depth.get_height();
  % retrieve UINT16 depth vector
  depthVector = depth.get_data();
  % reshape vector, and scale to depth in meters
  depthMap = double(transpose(reshape(depthVector, [depthWidth,depthHeight]))) .* depthScale;
  %align the depth frames to the color stream
  aligned_frames = alignedFs.process(fs);
  depth = aligned_frames.get_depth_frame(); 
  %Select rgb frame
  color = fs.get_color_frame();
  colordata=color.get_data();
  img = permute(reshape(colordata',[3,color.get_width(),color.get_height()]),[3 2 1]);
  subplot(2,1,1);
  imshow(depthMap)  
  impixelinfo;
  title('Depth Map (m)');
  subplot(2,1,2);
  imshow(img)
  impixelinfo;
  title('Color Map');
%end
% Stop streaming
  pipe.stop();
[image]

Please tell me what should I do now to allign the two frames so that I can use the depth value in RGB frames to clip the background. I am displaying the depth frame in distance and not using the colorizer.

Ahad19931, 2019-06-24:
Is there anyone who can help me in this regard?

rcohe032, 2019-07-05:
I just changed the order of your code and it worked. After the for loop you do not need 
"depth = fs.get_depth_frame();" because this is the unaligned data and it is not used. 
Right after the for loop put this first: 

`aligned_frames = alignedFs.process(fs);
depth = aligned_frames.get_depth_frame();`

and then you can continue with your code as is after moving that up. You were calculating the depthVector with the unaligned data, so you were never actually using the aligned depth frame anywhere. 

[image]
[image]

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

@rcohe032 Thanks for the help!
@Ahad19931 Did you get chance to try as rcohe032 suggested? Any other questions about this? Thanks!

RealSenseCustomerSupport, 2019-08-12:
----------------------------------------------

@Ahad19931 If no other questions, will close the ticket accordingly.

closed 2019-08-12 by RealSenseCustomerSupport