realsenseai/librealsense #4247: How to Calibrate T265 Camera?

zssjh, 2019-06-19:
As the title suggests， thanks~

MartyG-RealSense, 2019-06-19:
The documentation for the T265 states that it does not require calibration, as it is done on the production line.  The documentation has information on wheel odometry calibration though.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#wheel-odometry-calibration-file-format

You may also be interested in an example program for using the camera intrinsics and extrinsics API for the T265 that was introduced in RealSense SDK version 2,.20.0

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic

dorodnic, 2019-06-24:
The T265 comes precalibrated and does not require any user recalibration. If you want to integrate it with 3rd-party wheel-odometry you need to provide relevant extrnsics.

closed 2019-06-24 by dorodnic

GigaFlopsis, 2019-11-27:
However, I have found that the pre-calibration parameter is incorrect.  This did not affect the quality of navigation, but I need to use other calibration parameters to use the camera.

neilyoung, 2019-12-19:
@GigaFlopsis 

> However, I have found that the pre-calibration parameter is incorrect. This did not affect the quality of navigation

Would you mind to share, what problems you see with the current calibration? I'm also thinking the calibration is not optimal, since the drift is is pretty high at large scale, especially outdoors.

GigaFlopsis, 2019-12-20:
I'm only talking about calibrating the camera. Surprisingly, the navigation in the room works perfectly. I also compared the navigation with halens and it is similar. If we talk about navigation on the street, I have also had problems, I suspect that this is due to the small number of features at a short distance.

toolagu, 2020-03-19:
I also want to know how to calibrate the t265 imu. I have 2 t265, one is work well and one not . the not work one when I go ahead 1m, it walked forward 1 meter to the left (30 degrees from my positive direction，another good one shows right . how can I calibrate the imu ??

skylouis, 2020-09-08:
Hi every-one, 
After my T265 had performed a few drops the results of the slam algorithm wasn't accurate. I think the imu calibration is not valid anymore.
@dorodnic  is there way to re-calibrate the imu ?

octavio2895, 2021-05-10:
Im also having issues with the T265 IMU calibration, it reports 9.92m/s^2 when completely vertical and a magnitude of over 10 when stationary. This causes the camera to dip ~0.40m below ground over ~15m. Its interfering with the navigation stack.

AncientRemember, 2021-05-14:
anyone calibrate the camera parameters,is it accurate?

Max-H-JJ, 2021-07-21:
I have tried to calibrate the T265 camera focusing on single fisheye,  with the **realsense** python camera rectify tuorial to verify my calibration results. However, no matter how many times I tried, the computed parameters and matrix cannot fully meet the product pre-inserted ones, which means I cannot undistort and rectify the captured images. 

**This is the captured image, and the following is the rectified image using my own calibrated parameters.**
[image]
[image]