realsenseai/librealsense #4252: rs2_delete_pipeline_profile not return, unable to stop

wbing-xu, 2019-06-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | newest | 
| Operating System & Version |   Android8.1  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | RK3399 |
| SDK Version                            | 2.23}                          | 
| Language                            | c++                    | 
| Segment			|  Robot }                          | 

### Issue Description

When I repeatedly open and close the device, occasionally a stop failure occurs.

The final error log:
[image]

My code looks like this:
```
void doing()
{
    struct RealSenseInfo
    {
        std::unique_ptr<rs2::pipeline> pipeline_;
        rs2::pipeline_profile profile_;
    };
    std::map<std::string, RealSenseInfo> serial_realsense_info_;

    rs2::context ctx;

    while (true)
    {
        //open device
        for (auto &&dev : ctx.query_devices())
        {
            dev.hardware_reset();

            auto advanced_dev = dev.as<rs400::advanced_mode>();
            advanced_dev.first<rs2::depth_sensor>().set_option(rs2_option::RS2_OPTION_VISUAL_PRESET, rs2_rs400_visual_preset::RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);

            rs2::config cfg;
            auto serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
            auto physical_pport = dev.get_info(RS2_CAMERA_INFO_PHYSICAL_PORT);
            cfg.enable_stream(RS2_STREAM_DEPTH, width_, height_, RS2_FORMAT_Z16, fps_);
            cfg.enable_stream(RS2_STREAM_INFRARED, 1, width_, height_, RS2_FORMAT_Y8, fps_);
            cfg.enable_device(serial);
            auto pipe = std::make_unique<rs2::pipeline>(ctx);
            auto profile = pipe->start(cfg);

            RealSenseInfo real_sense_info;
            real_sense_info.pipeline_ = std::move(pipe);
            real_sense_info.profile_ = std::move(profile);
            serial_realsense_info_.insert(pair<std::string, RealSenseInfo>(serial, std::move(real_sense_info)));
        }

        //do something
        {
            while (...)
            {
                for (auto &kv : serial_realsense_info_)
                {
                    //poll_frame.......
                }
            }
        }

        //stop device
        serial_realsense_info_.clear(); //here,occasionally not return
    }
}
```

dorodnic, 2019-06-22:
Looks like a bug to me...

RealSenseSupport, 2021-02-25:
There's been many [improvements ](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2420)for Android since 2.23. Please let us know if this issue still persists on SDK v2.42.0. Thank you.

RealSenseSupport, 2021-03-05:
Closing this ticket for now. Please let us know if the issue persists in the more recent releases. Thank you.

closed 2021-03-05 by RealSenseSupport